openwrt/packages #12394: [perlbase-cpan] missing dependency module Test2::Util

n8v8R, 2020-06-03:
Maintainer: @Naoir | @pprindeville 
Environment: all arch | treewide

Description: `install` implies test 

trying to install some module terminates with

>Can't locate Test2/Util.pm in @INC (you may need to install the Test2::Util module)

that module https://metacpan.org/pod/Test2 is absent from the OpenWrt and without it `cpan` is basically useless as it creates a catch22 - the module cannot be installed via `cpan` and without the module installation of modules fails.

dl12345, 2020-06-04:
It is there. Can't remember exactly which package though. The only way I could get perl to work reliably was to install every single perl package in the repo. There are so many interdependencies...

n8v8R, 2020-06-04:
There is *perlbase-test* in the repo but that does not cover it.

*Test2* source code seems to be located here https://github.com/Test-More/test-more/ and I ran a search on the OpenWrt code base for that string `https://github.com/Test-More/test-more` as expecting such showing for `PKG_SOURCE_URL`. But that search comes up empty

dl12345, 2020-06-04:
@pprindeville might be able to throw some light on it

dl12345, 2020-06-04:
So, looking at the code it's actually there in the build tree: it's called Test-Simple: https://metacpan.org/release/Test-Simple, and it contains both Test which you refer to above and Test2.

Test2 is actually being built, but the perl makefiles install Test but don't install Test2, which are both located in the same part of the build tree.

So the package Makefile would need to be modified to install it. Also perl also isn't compiled by default with the test option checked under Configuration, which is required for installation of Test

pprindeville, 2020-08-09:
Sounds like most of the work is done.

Do you have a PR for me? :smirk:

closed 2020-09-07 by n8v8R