openwrt/packages #12403: PHP: segmentation fault when exiting

barhom, 2020-06-04:
When running any PHP script the script finishes successfully however exits with a "Segmentation fault".

Running master from 2020-06-04.
Packages installed:
```
opkg list-installed | grep php
php7 - 7.4.6-1
php7-cli - 7.4.6-1
php7-mod-filter - 7.4.6-1
php7-mod-json - 7.4.6-1
php7-mod-mbstring - 7.4.6-1
php7-mod-openssl - 7.4.6-1
```
The php script:
```
# cat test.php 
<?php
echo "HELLO WORLD\n";
```

The output:
```
# php-cli test.php 
HELLO WORLD
Segmentation fault
```

barhom, 2020-06-06:
@mhei do you know what is causing this?

Thanks

mhei, 2020-06-14:
I traced it down with gdb to be a problem in oniguruma:

```
Program received signal SIGSEGV, Segmentation fault.
0xb6e37f70 in i_free_callout_name_entry (key=0x369040, e=0x0, arg=0x37afe0) at oniguruma-6.9.5_rev1/src/regparse.c:1297
```

I'll create a PR with a fix, see https://github.com/kkos/oniguruma/pull/196.

PR #12536 (2020-06-18): oniguruma: build using autotools
Maintainer: me
Compile tested: mvebu, WRT3200ACM, openwrt master
Run tested: as above, running php-cli with a "hello world" script

Description:
Cmake currently fails to compile properly.  While there's a fix upstream, we can avoid carrying any patches here, and use the more stable build system, despite being slower.

---

I thought about using https://github.com/kkos/oniguruma/commit/c07fa0aeae09ea7a1564539de92dd9b5ae078592 to keep using cmake, but I feel we're jumping through hoops to do it.
I know autotools uses more resources to build, so I'm not committing [... cut]
closed 2020-06-18 by cotequeiroz