openwrt/packages #12411: zstd: build failure

dl12345, 2020-06-05:
```
user@ubuntu:/opt/openwrt$ make package/zstd/compile V=s
make[1]: Entering directory '/opt/openwrt'
make[2]: Entering directory '/opt/openwrt/feeds/packages/devel/ninja'
make[2]: Leaving directory '/opt/openwrt/feeds/packages/devel/ninja'
time: package/feeds/packages/ninja/host-compile#0.18#0.03#0.19
make[2]: Entering directory '/opt/openwrt/feeds/packages/devel/meson'
make[2]: Leaving directory '/opt/openwrt/feeds/packages/devel/meson'
time: package/feeds/packages/meson/host-compile#0.15#0.04#0.17
make[2]: Entering directory '/opt/openwrt/package/libs/toolchain'
echo "libc" >> /opt/openwrt/staging_dir/target-x86_64_glibc/pkginfo/toolchain.default.install
echo "libgcc" >> /opt/openwrt/staging_dir/target-x86_64_glibc/pkginfo/toolchain.default.install
echo "libatomic" >> /opt/openwrt/staging_dir/target-x86_64_glibc/pkginfo/toolchain.default.install
echo "libstdcpp" >> /opt/openwrt/staging_dir/target-x86_64_glibc/pkginfo/toolchain.default.install
echo "libpthread" >> /opt/openwrt/staging_dir/target-x86_64_glibc/pkginfo/toolchain.default.install
echo "librt" >> /opt/openwrt/staging_dir/target-x86_64_glibc/pkginfo/toolchain.default.install
echo "ldconfig" >> /opt/openwrt/staging_dir/target-x86_64_glibc/pkginfo/toolchain.default.install
make[2]: Leaving directory '/opt/openwrt/package/libs/toolchain'
time: package/libs/toolchain/compile#0.16#0.02#0.17
make[2]: Entering directory '/opt/openwrt/feeds/packages/utils/zstd'
mkdir -p /opt/openwrt/dl
SHELL= flock /opt/openwrt/tmp/.zstd-1.4.5.tar.zst.flock -c '  	/opt/openwrt/scripts/download.pl "/opt/openwrt/dl" "zstd-1.4.5.tar.zst" "2c2366874bc449ff539614266d8c0d6ecdb4baf30bb65609c239ab4ed23c03c7" "" "https://github.com/facebook/zstd/releases/download/v1.4.5"    '
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://sources.cdn.openwrt.org/zstd-1.4.5.tar.zst
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 1403k  100 1403k    0     0  6022k      0 --:--:-- --:--:-- --:--:-- 6022k
touch /opt/openwrt/build_dir/target-x86_64_glibc/zstd-1.4.5/.prepared_44363525c9c004d1e2fd0627801f0355_6664517399ebbbc92a37c5bb081b5c53_check
. /opt/openwrt/include/shell.sh;  /opt/openwrt/dl/zstd-1.4.5.tar.zst | tar -C /opt/openwrt/build_dir/target-x86_64_glibc/zstd-1.4.5/.. -xf -
bash: /opt/openwrt/dl/zstd-1.4.5.tar.zst: Permission denied
tar: This does not look like a tar archive
tar: Exiting with failure status due to previous errors
make[2]: *** [Makefile:105: /opt/openwrt/build_dir/target-x86_64_glibc/zstd-1.4.5/.prepared_44363525c9c004d1e2fd0627801f0355_6664517399ebbbc92a37c5bb081b5c53] Error 2
make[2]: Leaving directory '/opt/openwrt/feeds/packages/utils/zstd'
time: package/feeds/packages/zstd/compile#0.18#0.05#0.41
make[1]: *** [package/Makefile:113: package/feeds/packages/zstd/compile] Error 2
make[1]: Leaving directory '/opt/openwrt'
make: *** [/opt/openwrt/include/toplevel.mk:224: package/zstd/compile] Error 2
```

So it looks like the untar command needs to be modified to use `tar -I zstd -xvf` on the archive, or switch to the `.tar.gz` version

My host is ubuntu 20.04

cc: @neheb

neheb, 2020-06-05:
sounds like an SDK problem.

dl12345, 2020-06-05:
There was a commit from a few days ago in the main repo that updated `include/unpack.mk` to add the zstd unpack method. I hadn't merged that commit yet, so the command which should have read 

`. /opt/openwrt/include/shell.sh;  zstdcat /opt/openwrt/dl/zstd-1.4.5.tar.zst | tar -C /opt/openwrt/build_dir/target-x86_64_glibc/zstd-1.4.5/.. -xf -` 

instead had the `zstdcat` string blank and read 

`. /opt/openwrt/include/shell.sh;  /opt/openwrt/dl/zstd-1.4.5.tar.zst | tar -C /opt/openwrt/build_dir/target-x86_64_glibc/zstd-1.4.5/.. -xf -` 

so it was trying to execute an archive file; problem fixed.

closed 2020-06-05 by dl12345