openwrt/packages #12413: Adblock TypeError fs.exec_direct is not a function

feh123, 2020-06-05:
Hi I am running OpenWrt 19.07.0 r10860 on a RPI +3B. I have an Alfa USB wifi and configure the onboard wifi as the client and the Alfa wifi as Master. I have openvpn installed - the unit is working very well as an AP - openwrt is very good. 
I wanted to use Adblock as well. I have tried twice now to follow the openwrt instructions as install Adblock via Luci. The installation goes without a problem but I get a typeerror when I open Adblock. The text that follow is in the subject line above. 
The Overview page is blank.
The system log shows a series of errors like:  download of 'adguard' failed, download of yoyo failed. Otherwise I cannot see anything else wrong. But the unit still works as an AP and openvpn is working.
Any advice much appreciated.

neheb, 2020-06-06:
ping @dibdot

dibdot, 2020-06-06:
If I remember correctly, the read_direct() and exec_direct() functions has been backported to later 19.07.x release series. At least you have to update your luci base components as well. Said that, probably it's a much better approach to update the whole system to latest stable 19.07.3.

Regarding your download errors please doublecheck that you've installed ssl support and the ca-bundle package, as adblock checks all ssl certifiactes by default.

closed 2020-06-06 by dibdot

feh123, 2020-06-06:
@dibdot thanks - I updated to 19.07.3 and made sure the ssl support was there and that worked. Thanks for the advice.