openwrt/packages #12418: htop is missing kernel options for I/O and it shows no perm

BKPepe, 2020-06-06:
Maintainer: @champtar
Environment: all, OpenWrt 19.07 and OpenWrt master

Description:

If you would like to know I/O rate of read/write for processes which helps you to know which processes are writing it to flash memory, it is necessary to enable these kernel configs:
- [IO_ACCOUNTING](https://cateee.net/lkddb/web-lkddb/TASK_IO_ACCOUNTING.html)
- [TASK_DELAY_ACCT](https://cateee.net/lkddb/web-lkddb/TASK_DELAY_ACCT.html)

Both kernel options are not set.
https://github.com/openwrt/openwrt/blob/openwrt-19.07/target/linux/generic/config-4.14#L4875
https://github.com/openwrt/openwrt/blob/openwrt-19.07/target/linux/generic/config-4.14#L4873

And this is same for OpenWrt master:
https://github.com/openwrt/openwrt/blob/master/target/linux/generic/config-5.4#L5474
https://github.com/openwrt/openwrt/blob/master/target/linux/generic/config-5.4#L5472

Otherwise, htop shows you `no perm`.
Related discussion: http://forum.tinycorelinux.net/index.php?topic=23161.0

[image]

This prevents to fully use htop. Can you please enable it in OpenWrt main repository?

n8v8R, 2020-06-22:
also required for full *sysstat* functionality

jow-, 2020-06-22:
> Can you please enable it in OpenWrt main repository?

Whats the effect on the kernel image size?

n8v8R, 2020-07-24:
@jow- ~5kB

wulfy23, 2020-11-20:
so thats what all those noperm's are! thanks BK...

DocMAX, 2023-04-08:
i enabled both in kernel 5.15 but still says "N/A"

peci1, 2023-04-08:
Do you run htop with root privileges?

DocMAX, 2023-04-08:
yes of course, there is no other than root on openwrt. i now added and compiling with these extra lines:
```
CONFIG_KERNEL_TASK_DELAY_ACCT=y
CONFIG_KERNEL_TASK_IO_ACCOUNTING=y
```