openwrt/packages #12420: darkstat: running darkstat will make ssh-ing to device impossible.

rezad1393, 2020-06-06:
Maintainer: @padre-lacroix 
Environment:
Linux openwrt 4.14.180 #0 SMP Sat May 16 18:32:20 2020 armv7l GNU/Linux
wrt1200 linksys

Description:
running darkstat makes the ssh-ing to device impossible.
the error is this is logs:
for client:
```
kex_exchange_identification: read: Connection reset by peer
Connection reset by 192.168.1.1 port 22
```
for openwrt host:
`openwrt sshd[]: fatal: /var/empty must be owned by root and not group or world-writable.`

it seems that running darkstat makes /var/empty owned by nobody but sshd needs it to be owned by root.

fortunately I had an ssh running before installing darkstat because even removing darkstat doesn't fix the issue and it need manual chown.

this happens every time darkstat is run.

padre-lacroix, 2020-06-07:
I looked into that and effectively darkstat makes the directory /var/empty owned by nobody.  The command for that is in the init script of darkstat in /etc/init.d.
On my router, where I have darkstat running, I have no problem to log in by SSH, therefore I do not see the problem that is created.
In the init script, there is a line that can be commented out:
set_config_string(){
	mkdir -p $RUN_D
#	chown $USER:$GROUP $RUN_D
	. /lib/functions/network.sh
	config_load $CONFIGNAME
	config_foreach build_config_string darkstat

Can you try to comment that line (chown $USER:$GROUP $RUN_D) and let me know if that works.  You may have to delete the directory /var/empty first.

The /var/empty is used by darkstat to store the pid file, and darkstat creates the directory.  Another option would be to create the dirtectory /var/empty/darkstat instead of /var/empty: to do that, leave the line
chown $USER:$GROUP $RUN_D
uncommented and change what $RUN_D is at the beginning of the darkstat file in /etc/init.d:
RUN_D=/var/empty/darkstat.

Would it be possible to test these to see if any of these items would work and let SSH works also.  As I said, my router does not have the problem with SSH and therefore I cannot test whether or not SSH is prevented.

padre-lacroix, 2020-06-07:
I tested both options in the file /etc/init.d/darkstat and both seem to be working as far as darkstat is concerned:
Either:
1- comment the line
chown $USER:$GROUP $RUN_D

or
2- set the RUN_D variable to RUN_D=/var/empty/darkstat

In both cases, the pid file is created correctly and removed when darkstat is closed.

As far as SSH is involved, I cannot test it as I do not see the problem with SSH.

neheb, 2020-06-07:
He might be using opemssh, not dropbear.

rezad1393, 2020-06-07:
I am sorry I didnt mention the openssh.
yes I use openssh instead dropbear.
but I thought that this line would clear that I am using openssh

` openwrt **sshd**[]: fatal: /var/empty must be owned by root and not group or world-writable.`

should I just use dropbear?
I use openssh because I can use sftp easily and also because of openssh option that I use to make it more secure (as a noob just copy pasting from the web)

padre-lacroix, 2020-06-11:
I changed a little the approach that I will be using in the file /etc/init.d/darkstat.
First, the RUN_D variable will be set to RUN_D=/var/darkstat (which correspond to /tmp/darkstat) and
Second: the line
chown $USER:$GROUP $RUN_D
will be removed.

Darkstat is working fine that way, and there will be no possible conflict with sshd.
I will try to have a pull request for this over the weekend: it has been a while since I did one and I do not remember exactly how to do it (but it will come back).

closed 2020-09-19 by neheb