openwrt/packages #12443: ztsd: meson-based build requires `python3-distutils` presence in host system

ryzhovau, 2020-06-07:
Can be reproduced on clean Debian 10 with [Buildroot requirements](https://oldwiki.archive.openwrt.org/doc/howto/buildroot.exigence#install_procedure_on_linux) installed.

``
$ make V=s package/zstd/compile
...
The Meson build system
Version: 0.54.2
Source dir: /home/ryzhovau/Entware/build_dir/target-mipsel_mips32r2_glibc-2.31/zstd-1.4.4/build/meson
Build dir: /home/ryzhovau/Entware/build_dir/target-mipsel_mips32r2_glibc-2.31/zstd-1.4.4/build/meson/openwrt-build
Build type: cross build
Using 'PKG_CONFIG_PATH' from environment with value: '/home/ryzhovau/Entware/staging_dir/target-mipsel_mips32r2_glibc-2.31/opt/lib/pkgconfig:/home/ryzhovau/Entware/staging_dir/target-mipsel_mips32r2_glibc-2.31/opt/share/pkgconfig'
WARNING: Unknown options: "bin_control"
The value of new options can be set with:
meson setup <builddir> --reconfigure -Dnew_option=new_value ...
Project name: zstd
Project version: DUMMY
C compiler for the build machine: gcc (gcc 8.3.0 "gcc (Debian 8.3.0-6) 8.3.0")
C linker for the build machine: gcc ld.bfd 2.31.1
C++ compiler for the build machine: g++ (gcc 8.3.0 "g++ (Debian 8.3.0-6) 8.3.0")
C++ linker for the build machine: g++ ld.bfd 2.31.1
Using 'CFLAGS' from environment with value: ''
Using 'LDFLAGS' from environment with value: ''
C compiler for the host machine: mipsel-openwrt-linux-gnu-gcc (gcc 8.4.0 "mipsel-openwrt-linux-gnu-gcc (OpenWrt GCC 8.4.0 r1223-5ceb6995) 8.4.0")
C linker for the host machine: mipsel-openwrt-linux-gnu-gcc ld.bfd 2.31.1
Using 'LDFLAGS' from environment with value: ''
C++ compiler for the host machine: mipsel-openwrt-linux-gnu-g++ (gcc 8.4.0 "mipsel-openwrt-linux-gnu-g++ (OpenWrt GCC 8.4.0 r1223-5ceb6995) 8.4.0")
C++ linker for the host machine: mipsel-openwrt-linux-gnu-g++ ld.bfd 2.31.1
Build machine cpu family: x86_64
Build machine cpu: x86_64
Host machine cpu family: mips
Host machine cpu: generic
Target machine cpu family: mips
Target machine cpu: generic
Program python3 found: YES (/home/ryzhovau/Entware/staging_dir/host/bin/python)

../../../../build_dir/target-mipsel_mips32r2_glibc-2.31/zstd-1.4.4/build/meson/meson.build:25:0: ERROR: <ExternalProgram 'python3' -> ['/home/ryzhovau/Entware/staging_dir/host/bin/python']> is not a valid python or it is missing setuptools

A full log can be found at /home/ryzhovau/Entware/build_dir/target-mipsel_mips32r2_glibc-2.31/zstd-1.4.4/build/meson/openwrt-build/meson-logs/meson-log.txt
```

The reason is `$(STAGING_DIR_HOSTPKG)/bin/python3`, which tries to import distutils from host system:
```
$ cat /home/ryzhovau/Entware/build_dir/target-mipsel_mips32r2_glibc-2.31/zstd-1.4.4/build/meson/openwrt-build/meson-logs/meson-log.txt
...
---
Sanity check compile stderr:

-----
C++ compiler for the host machine: mipsel-openwrt-linux-gnu-g++ (gcc 8.4.0 "mipsel-openwrt-linux-gnu-g++ (OpenWrt GCC 8.4.0 r1223-5ceb6995) 8.4.0")
C++ linker for the host machine: mipsel-openwrt-linux-gnu-g++ ld.bfd 2.31.1
Build machine cpu family: x86_64
Build machine cpu: x86_64
Host machine cpu family: mips
Host machine cpu: generic
Target machine cpu family: mips
Target machine cpu: generic
Program python3 found: YES (/home/ryzhovau/Entware/staging_dir/host/bin/python)
Could not introspect Python (['/home/ryzhovau/Entware/staging_dir/host/bin/python', '-c', "import sysconfig\nimport json\nimport sys\n\ninstall_paths = sysconfig.get_paths(scheme='posix_prefix', vars={'base': '', 'platbase': '', 'installed_base': ''})\n\ndef links_against_libpython():\n    from distutils.core import Distribution, Extension\n    cmd = Distribution().get_command_obj('build_ext')\n    cmd.ensure_finalized()\n    return bool(cmd.get_libraries(Extension('dummy', [])))\n\nprint (json.dumps ({\n  'variables': sysconfig.get_config_vars(),\n  'paths': sysconfig.get_paths(),\n  'install_paths': install_paths,\n  'version': sysconfig.get_python_version(),\n  'platform': sysconfig.get_platform(),\n  'is_pypy': '__pypy__' in sys.builtin_module_names,\n  'link_libpython': links_against_libpython(),\n}))\n"]): exit code 1
Program stdout:

Program stderr:

Traceback (most recent call last):
  File "<string>", line 20, in <module>
  File "<string>", line 8, in links_against_libpython
ModuleNotFoundError: No module named 'distutils.core'

../../../../build_dir/target-mipsel_mips32r2_glibc-2.31/zstd-1.4.4/build/meson/meson.build:25:0: ERROR: <ExternalProgram 'python3' -> ['/home/ryzhovau/Entware/staging_dir/host/bin/python']> is not a valid python or it is missing setuptools
```
Installation of distutils to host system by `apt-get install python3-distutils` solves thih problem.

Maintainer: @neheb, @commodo
Environment: Entware repo, mipsel feed.

neheb, 2020-06-07:
meson for 19.07 requires pyrhon3/host.

jefferyto, 2020-06-07:
`../staging_dir/host/bin/python` is build host Python, not python3/host (`$(STAGING_DIR_HOSTPKG)/bin/python3`).

I think this error is because a fresh Debian install has a minimal install of Python (last time I checked the package was `python3.7-minimal`), which has a minimal version of distutils. (I noted a [possible issue with meson](https://github.com/openwrt/packages/pull/11843#issuecomment-613493308) before, based on [investigating an issue with gpsd/scons](https://github.com/openwrt/packages/issues/10993#issuecomment-574405236).)

As you discovered, installing the full distutils package is one way to work around the issue. Having meson depend on python3/host is another way, though I suppose some documentation saying "meson on Debian requires python3-distutils" is another solution.

ryzhovau, 2020-06-07:
I just wonder: if meson trying to use python3/host, why python3/host is trying to use modules from host system instead of `$(STAGING_DIR_HOSTPKG)/lib/python$(PYTHON3-VERSION)`?

That's why I mentioned @commodo and/or @jefferyto :)

neheb, 2020-06-07:
Oh this is a Debian problem.

neheb, 2020-06-07:
Soooo...should we switch to use python3/host as in 19.07?

@ryzhovau https://github.com/openwrt/packages/commit/125f6ff9ec01c57e0cf57b042b95884f160cf48f is applied to 19.07 only. 19.07 base only supports python 2, so python3/host is needed.

jefferyto, 2020-06-08:
@neheb I wonder if meson/zstd can be patched to use the minimal distutils.

@ryzhovau meson in 19.07 uses python3/host, meson in master doesn't (this was a deliberate choice of the maintainer @dhewg).

neheb, 2020-06-08:
Beats me. I have no idea what’s involved.

dhewg, 2020-06-08:
`include/prereq-build.mk` checks for python3 already, so it's a hard dependency already. And meson officially support running it directly from the source directory. I just combined those two facts instead of dragging in a rather big dependency.

debian has `python3-minimal` and `python3`. Only the latter depends on `libpython3-stdlib`, so it's not just distutils, the whole standard library is missing from the minimal package. Quite some ports run python scripts as part of their build process, so I guess zstd is not the only package running into this? If not the whole image build process, `scripts/json_add_image_info.py` comes to mind.

Instead of adding python3/host I think `include/prereq-build.mk` should be extended to check for stdlib.

And if I remember correctly the 19.07 branch depends on python3/host because of python2 vs 3?

jefferyto, 2020-06-08:
`python3.8-minimal` [depends on](https://packages.debian.org/bullseye/python3.8-minimal) `libpython3.8-minimal`.

Searching [openwrt-devel](https://duckduckgo.com/?q=openwrt-devel+debian+python+site%3Alists.infradead.org) and [Flyspray](https://bugs.openwrt.org/index.php?string=debian+python&project=2&do=index&type%5B%5D=&sev%5B%5D=&pri%5B%5D=&due%5B%5D=&reported%5B%5D=&cat%5B%5D=&status%5B%5D=open&percent%5B%5D=&opened=&dev=&closed=&duedatefrom=&duedateto=&changedfrom=&changedto=&openedfrom=&openedto=&closedfrom=&closedto=) doesn't appear to return any relevant results; presumably if image build doesn't work on Debian someone would have reported it.

dhewg, 2020-06-08:
Right you are, missed that tail dependency...

But still:
```
$ grep -w import scripts/json_add_image_info.py
from os import getenv
from pathlib import Path
from sys import argv
import hashlib
import json

$ dpkg -L libpython3.8-minimal:amd64|grep json   

$ dpkg -L libpython3.8-stdlib:amd64|grep json
/usr/lib/python3.8/json
/usr/lib/python3.8/json/__init__.py
/usr/lib/python3.8/json/decoder.py
/usr/lib/python3.8/json/encoder.py
/usr/lib/python3.8/json/scanner.py
/usr/lib/python3.8/json/tool.py
/usr/lib/python3.8/lib-dynload/_json.cpython-38-x86_64-linux-gnu.so
```
That shouldn't work with just -minimal due to the missing json, right?

jefferyto, 2020-06-09:
Yes, that script should fail at `import json`. It looks like that script was added last year (openwrt/openwrt@881ed09ee6e23f6c224184bb7493253c4624fb9f) to help generate additional JSON files, which is not critical to the image building process.

I don't have a strong opinion on whether a full Python installation should be required for the OpenWrt buildroot. You are free to ask or propose a patch to openwrt-devel. My feeling is that folks there will push back against "dragging in a rather big dependency" and instead opt to rewrite the few Python scripts so they do not need a full stdlib, but that's just my opinion.

dhewg, 2020-06-09:
I'm just trying to avoid having meson depend on python3/host, and this makes me think it is avoidable:
https://mesonbuild.com/Use-of-Python.html#dependency-hell

But all this still sounds like a debian specific packaging issue to me, afaiu there is no 'minimal python'.

Anyway, it looks like the issue is because of this line:
https://github.com/facebook/zstd/blob/dev/build/meson/meson.build#L25

And this sounds like the very same issue:
https://github.com/mesonbuild/meson/pull/7027

Which in return sounds like it can be fixed with replacing this
```
python3 = import('python').find_installation()
GetZstdLibraryVersion_py = files('GetZstdLibraryVersion.py')
r = run_command(python3, GetZstdLibraryVersion_py, zstd_h_file)
```
with this
```
find_program('GetZstdLibraryVersion.py', native : true)
r = run_command(python3, GetZstdLibraryVersion_py, zstd_h_file)
```

for the zstd package

neheb, 2020-06-09:
Surprising. I assume someone wants to make a PR? :)

dhewg, 2020-06-09:
Please feel free, I'm about to leave for vacation and can't test this short term...

neheb, 2020-06-09:
Just tested on a debian 10 VM. Works.

closed 2020-06-10 by neheb