openwrt/packages #12444: unbound: config "domain" is lost on restart

AndreaBorgia-Abo, 2020-06-07:
Maintainer: Eric Luehrsen ericluehrsen@gmail.com
Environment: Fritzbox 4040, OpenWRT 19.07.3

Description:
1. configure unbound in parallel mode
2. insert [relevant "domain" keyword](https://github.com/openwrt/packages/commit/b6caafef46829922dade557d6b3082360053bf8f) for custom domain in /etc/config/unbound
3. restart unbound
4. verify that resolv.conf now has a "search lan." line instead of "search custodomain."
5. verify that the keyword "domain" has disappeared from the config

Issue first discussed in the [forum](https://forum.openwrt.org/t/unbound-option-domain-is-lost/64987/12?u=aboaboit)

EricLuehrsen, 2020-06-13:
This is a LuCI page, tab, and work-flow apply issue. Please close this issue and copy it to openwrt/luci repository. I have not dug deeply into it yet. It might be simple or not. This issue in the right place will help receive the right help if I need it.

AndreaBorgia-Abo, 2020-06-14:
Done, thank you.

closed 2020-06-14 by AndreaBorgia-Abo