openwrt/packages #12470: [net / unbound-daemon-heavy] features not compiled in

n8v8R, 2020-06-10:
Maintainer: @EricLuehrsen 
Environment: OpenWrt tree wide

Description: not compiled into the daemon

* subnetcache
* dnscrypt

Since this being "heavy" variant could we please have those compiled in as well, adding to

`ifeq ($(BUILD_VARIANT),heavy)`

+ --enable-subnet
+ --enable-dnscrypt

EricLuehrsen, 2020-06-13:
I have found that packing a heavy and light version of Unbound is just not working. Between the daemon, library, and utilities and opkg simplistic dependencies, it has been a mess. I think it will change to up-select or down-select options in `menuconfig`. A "fair default" single package will be built for package binaries. Special cases will need to use image builder or compile all from source. The python module is a certain "no" for default option, because then python becomes a dependency. Its huge and not just a little heavier. A theoretical security risk must be considered for such powerful script language attached to a system necessary server.

n8v8R, 2020-06-14:
Thanks for the feedback. Too bad for py but understandable. *subnetcache* would be nice to have.

closed 2020-06-14 by n8v8R

EricLuehrsen, 2020-06-24:
Would you try #12580?

n8v8R, 2020-07-01:
After having removed all previous *unbound-** stuff and installed _unbound-daemon_ 1.10.1-4 instead neither the _subnetcahce_ or the _python_ modules are working.

```
unbound -V
Version 1.10.1

Configure line: --target=arm-openwrt-linux --host=arm-openwrt-linux --build=x86_64-pc-linux-gnu --program-prefix= --program-suffix= --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/lib --sysconfdir=/etc --datadir=/usr/share --localstatedir=/var --mandir=/usr/man --infodir=/usr/info --disable-nls --disable-dsa --disable-gost --enable-allsymbols --enable-ecdsa --enable-tfo-client --enable-tfo-server --with-libexpat=/builder/shared-workdir/build/sdk/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/usr --with-ssl=/builder/shared-workdir/build/sdk/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/usr --with-user=unbound --with-run-dir=/var/lib/unbound --with-conf-file=/var/lib/unbound/unbound.conf --with-pidfile=/var/run/unbound.pid --enable-ipset --with-libmnl=/builder/shared-workdir/build/sdk/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/usr --enable-event-api --with-libevent=/builder/shared-workdir/build/sdk/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/usr --with-pthreads
Linked libs: pluggable-libevent 2.1.11-stable (it uses epoll), OpenSSL 1.1.1g  21 Apr 2020
Linked modules: dns64 ipset respip validator iterator
TCP Fastopen feature available
```

EricLuehrsen, 2020-07-02:
python will not work with the UCI right now. You need to use manual configuration and do not chroot. python needs to a have a directory bind alias within /var/lib/unbound to make that happen.

n8v8R, 2020-07-02:
> python, subnetcache, and dnscrypt are not compiled into the default package. These are compile on your own options which are easily selected in `make menuconfig` using OpenWrt make environment.

That been a misconception of the patch set on my part then. Basically to get the functionality one has to the self-compile.

Other than that the new unified package works thus far with no issues observed, that is with manual config (as opposed through UCI for which I am having a hard to figure out which UCI settings translate to which unbound settings, aside from not scattering the config throughout various files)

PR #12581 (2020-06-27): unbound: reduced to one package with selectable options
Maintainer: me
Compile tested: Linksys WRT3200ACM all options
Run tested: Linksys WRT3200ACM with default or lighter options
Description: Enough is enough. Dual build Unbound and helper packages is a mess. One Unbound package will be presented with fair defaults. Up select or down select options under libraries/network/libunbound to enhance Unbound or squeeze Unbound into your desired target. Also add a new option to pop in helper DNS programs on the local host. I don't have use cases for the extended options. So I compiled them, but I do [... cut]