openwrt/packages #12482: zerotier: segfault on mvebu

ja-pa, 2020-06-11:
Maintainer: @mwarning 
Environment: Turris Omnia (TOS5), OpenWrt master
and
Environment: Turris Omnia (TOS5), OpenWrt 19.07

Description:
Zerrotier ends with seqfault

Step to reproduce:
```
$ zerotier-cli 
Segmentation fault
```

neheb, 2020-06-11:
What happens if you compile with libcxx?

ja-pa, 2020-06-12:
This is what I get with strace
```
strace zerotier-cli
execve("/usr/bin/zerotier-cli", ["zerotier-cli"], 0xbe9f0dc0 /* 14 vars */) = 0
set_tls(0xb6fdb5d8)                     = 0
set_tid_address(0xb6fdc1b0)             = 8738
open("/etc/ld-musl-armhf.path", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/lib/libminiupnpc.so.17", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/local/lib/libminiupnpc.so.17", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/lib/libminiupnpc.so.17", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fcntl64(3, F_SETFD, FD_CLOEXEC)         = 0
fstat64(3, {st_mode=S_IFREG|0644, st_size=40963, ...}) = 0
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\0\32\0\0004\0\0\0"..., 936) = 936
mmap2(NULL, 110592, PROT_READ|PROT_EXEC, MAP_PRIVATE, 3, 0) = 0xb6f3d000
mmap2(0xb6f56000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0x9000) = 0xb6f56000
close(3)                                = 0
open("/lib/libnatpmp.so.1", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/local/lib/libnatpmp.so.1", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/lib/libnatpmp.so.1", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fcntl64(3, F_SETFD, FD_CLOEXEC)         = 0
fstat64(3, {st_mode=S_IFREG|0755, st_size=8327, ...}) = 0
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0X\t\0\0004\0\0\0"..., 936) = 936
mmap2(NULL, 77824, PROT_READ|PROT_EXEC, MAP_PRIVATE, 3, 0) = 0xb6f2a000
mmap2(0xb6f3b000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0x1000) = 0xb6f3b000
close(3)                                = 0
open("/lib/libstdc++.so.6", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/local/lib/libstdc++.so.6", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/lib/libstdc++.so.6", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fcntl64(3, F_SETFD, FD_CLOEXEC)         = 0
fstat64(3, {st_mode=S_IFREG|0755, st_size=1056939, ...}) = 0
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\210\205\7\0004\0\0\0"..., 936) = 936
mmap2(NULL, 1130496, PROT_READ|PROT_EXEC, MAP_PRIVATE, 3, 0) = 0xb6e16000
mmap2(0xb6f21000, 36864, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0xfb000) = 0xb6f21000
mmap2(0xb6f29000, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0xb6f29000
close(3)                                = 0
open("/lib/libgcc_s.so.1", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fcntl64(3, F_SETFD, FD_CLOEXEC)         = 0
fstat64(3, {st_mode=S_IFREG|0644, st_size=40963, ...}) = 0
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0000E\0\0004\0\0\0"..., 936) = 936
mmap2(NULL, 110592, PROT_READ|PROT_EXEC, MAP_PRIVATE, 3, 0) = 0xb6dfb000
mmap2(0xb6e14000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0x9000) = 0xb6e14000
close(3)                                = 0
mprotect(0xb6f56000, 4096, PROT_READ)   = 0
mprotect(0xb6f3b000, 4096, PROT_READ)   = 0
mprotect(0xb6f21000, 20480, PROT_READ)  = 0
mprotect(0xb6e14000, 4096, PROT_READ)   = 0
--- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_ACCERR, si_addr=0x4f13c8} ---
+++ killed by SIGSEGV +++
Segmentation fault
```

mwarning, 2020-06-12:
Thank you for the strace, that is very helpful.

I see that mmap2 is called with null pointer as first parameter, which might be the cause for the crash. Further debugging would include a debug build of zerotier and gdb-server/gdb-client. But that might be a bit too much to ask for.

ja-pa, 2020-06-12:
Sorry, I wasn't able to create zerotier-cli binary with debug symbols even if I followed https://openwrt.org/docs/guide-developer/gdb

mwarning, 2020-06-13:
Zerotier has a debug option in "make menuconfig". That may do it.
Aynway, I wonder if I can get hold of a Turris.

ja-pa, 2020-06-13:
I tried that but end with following error
```
build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/ZeroTierOne-1.4.6/node/C25519.cpp:749:(.text+0x10954): dangerous relocation: unsupported relocation
collect2: error: ld returned 1 exit status
make-linux.mk:294: recipe for target 'zerotier-one' failed
make[3]: *** [zerotier-one] Error 1

```

c10l, 2020-07-15:
Is there any progress on this? Anything I can do to help?

mwarning, 2020-07-15:
I do not think so. Someone with access to a turris needs to debug the problem. E.g. using gdb.

BKPepe, 2020-07-16:
I would rather say that this issue is most likely related to Marvell Armada 38x, which is using Turris Omnia, ClearFrog, and so on. It appears that ZeroTier developers didn't ask people for more details. 

There's a possibility to use and tweak the proposed workaround from one of those issues to see if it fixes it or not. I agree that it is a hack, but at least something to get it working. Otherwise, as it was pointed out and more correct is to see something like `gdb`.

However, I see it very interesting that it was working on OpenWrt 18.06, but it does not work with OpenWrt 19.07.

neheb, 2020-07-16:
Compilers change.

Actually with GCC 10, this doesn’t even compile (internal compiler error). There’s a patch for that that I’ll submit here soon.

BKPepe, 2020-07-17:
Hm, I tried to narrow this issue a little bit down and it seems to be related somehow to ASLR PIE.  I added to Makefile `PKG_ASLR_PIE:=0` and zerotier works. It does no longer segfault on start.

mwarning, 2020-07-17:
@BKPepe nice! I've found this for more context: http://lists.infradead.org/pipermail/lede-commits/2018-January/005866.html
Do you want to make a PR?

BKPepe, 2020-07-17:
For now, I don't have any clue why it does not work with https://github.com/openwrt/packages/blob/master/net/zerotier/patches/0002-remove-pie.patch and in the meantime, I prepared this patch, which disables ASLR pie for zerotier: https://gitlab.nic.cz/turris/turris-build/-/blob/d71b9c1872f7deaffd15fc55717122e200714fd0/patches/packages/hack/0010-zerotier-disable-ASLR-PIE.patch , which I see as workaround/hack, whatever you prefer.

I will try to investigate it.

closed 2020-07-23 by neheb