openwrt/packages #12488: dawn still failing on arm64

neheb, 2020-06-11:
Please make sure that the issue subject starts with `<package-name>: `

Also make sure that the package is maintained in this repository and not in base which should be submitted at https://bugs.openwrt.org or in the LuCI repository which should be submitted at https://github.com/openwrt/luci/issues.

Issues related to releases below 18.06 and forks are not supported or maintained and will be closed.

# Issue template (remove lines from top till here)

Maintainer: @PolynomialDivision 
Environment: aarch64

https://downloads.openwrt.org/snapshots/faillogs/aarch64_cortex-a72/packages/dawn/compile.txt

This looks like a different error.

PolynomialDivision, 2020-06-12:
Could u look at this @dwmw2 ? :)

dwmw2, 2020-06-12:
Mea culpa; sorry.
```
src/utils/dawn_uci.c:44:46: error: field precision specifier '.*' expects argument of type 'int', but argument 4 has type 'size_t' {aka 'long unsigned int'} [-Werror=format=]
         snprintf(hostname, HOST_NAME_MAX, "%.*s", len, ptr.o->v.string);
                                            ~~^~   ~~~
```
That `len` wants to be a `size_t` instead of an `int`. Either change the variable and deal with any other issues, or just cast it in the line that causes the complaint. Will be at keyboard in an hour or so if you don't beat me to it.

PolynomialDivision, 2020-06-12:
> Mea culpa; sorry.
>
> ```
> src/utils/dawn_uci.c:44:46: error: field precision specifier '.*' expects argument of type 'int', but argument 4 has type 'size_t' {aka 'long unsigned int'} [...]

Yeah casting seems fine. I will do it by myself. thanks for looking at it.

commit 04d69bae (2020-06-12): dawn: update to latest version

Includes:
-  test_storage: further refactoring, added test functionality, and TESTING.md to describe testing approach
-  tcpsocket: implement client_read_cb with ustream_read my wheels are much worse than offical wheels.
-  Revert "tcpsocket: implement client_read_cb with ustream_read my wheels are much worse than offical wheels."
-  uci: fix arm64 compiling

Fixes: #12488
closed 2020-06-12 by dwmw2