openwrt/packages #12490: [utils/shadow] bcrypt should not be compiled in

n8v8R, 2020-06-12:
Maintainer: https://github.com/openwrt/openwrt/commit/9ba09653ade1fa20d0d777390d522fb179c1088e

Environment: tree wide

Description: `crypt_blowfish` is being removed in the toolchain https://git.openwrt.org/?p=openwrt/openwrt.git;a=blob;f=toolchain/musl/patches/901-crypt_size_hack.patch and therefore it does not make sense to compile `bcrypt` into the package https://github.com/openwrt/packages/blob/master/utils/shadow/Makefile#L44

neheb, 2020-06-14:
In that package, I adjust the default based on strength and what is available: https://github.com/openwrt/packages/blob/master/utils/shadow/Makefile#L131

You want https://github.com/openwrt/packages/blob/master/utils/shadow/Makefile#L44 to be changed based on availability?

n8v8R, 2020-06-14:
Checking on algo availability is more flexible and probably beneficial for users compiling their own kernels and retaining the algo.

neheb, 2020-06-16:
I can't find any reference to crypt_blowfish in shadow-utils.

n8v8R, 2020-06-16:
shadow does not provide the algo but the kernel

neheb, 2020-06-16:
Simple test:

```
passwd: failed to crypt password with salt '$2a$13$w8EJ0Yfz5bGsG4U/0m7bk/': Function not implemented
The password for root is unchanged.
```

Sounds like /etc/login.defs should be changed as needed.

n8v8R, 2020-06-16:
> edit: and no. shadow-utils does not use the kernel.

Yes, it does, https://forum.openwrt.org/t/passwd-a-sha256-sha512-less-secure-than-default/13117/3 and that is with `crypt_blowfis` being stripped from the kernel it does not work. If it would leverage any other crypto library, e.g. OpenSSL or WolfSSL salting with SHA|Blowfish would work.
___
> Sounds like /etc/login.defs should be changed as needed.

That is not an issue, except being left with DES or MD5 which are neither suitable for salting password hashes. 

https://openwrt-devel.openwrt.narkive.com/AtxRVHLp/sha512-sha256-hashes-truncated-busybox-weirdness

> All password hash algorithms except for MD5 (default) and DES (for compatibility reasons) have been stripped from libc to cut down on useless bloat.
>
>If you care about password storage security, it's better to use a function that was intentionally designed NOT to be fast, e.g. PBKDF2 or bcrypt.
___
Therefore compiling `bcrypt` into shadow is entirely pointless.

neheb, 2020-06-16:
I think the complete solution for OpenWrt is to port https://github.com/besser82/libxcrypt .

This is what fedora has done.

neheb, 2020-06-16:
Under glibc when manually modifying /etc/login.defs

```
passwd: failed to crypt password with salt '$2a$13$xbpmAYmq6Q/rZN5jOlNxJZ': Invalid argument
The password for root is unchanged.
```

neheb, 2020-06-16:
> > edit: and no. shadow-utils does not use the kernel.
> 
> Yes, it does, https://forum.openwrt.org/t/passwd-a-sha256-sha512-less-secure-than-default/13117/3 and that is why with `crypt_blowfish` being stripped from the kernel it does not work. If it would leverage any other crypto library, e.g. OpenSSL or WolfSSL salting with SHA|Blowfish would work.
That refers to musl the libc, not the kernel.
> 
> > Sounds like /etc/login.defs should be changed as needed.
> 
> That is not an issue, except being left with DES or MD5 which are neither suitable for salting password hashes.
Then porting libxcrypt is the solution for shadow-utils.
> https://openwrt-devel.openwrt.narkive.com/AtxRVHLp/sha512-sha256-hashes-truncated-busybox-weirdness
>
>
> Therefore compiling `bcrypt` into shadow is entirely pointless.
If libxcrypt is introduced here, that will not be true.

--without-bcrypt output in glibc:

```
Invalid ENCRYPT_METHOD value: 'BCRYPT'.
Defaulting to DES.
passwd: password changed.
```

n8v8R, 2020-06-16:
My bad about the mixup kernel vs glib.

The issue with those crypto offspring is code auditing  and long term maintenance but suppose since Fedora took a shine to it some auditing been done. 

Whether *libxcrypt* will make into OpenWrt remains to be seen.
___
> --without-bcrypt output in glibc:
> 
> ```
> Invalid ENCRYPT_METHOD value: 'BCRYPT'.
> Defaulting to DES.
> passwd: password changed.
> ```

That can be reverted to MD5 though defaulting to DES is really bad.
___
Anyway I reckon there is not need to spin this any further. In the current state compiling `bcrypt` into shadow makes no sense and bloats the code by a few bits, so leave it be then.

closed 2020-06-16 by n8v8R

neheb, 2020-06-16:
I spun up a PR fixing this anyway.

Fedora is financed by Red Hat. I assume they've done a fair amount of auditing given that they sell RHEL.