openwrt/packages #12495: mail() not available?

NemoAng, 2020-06-13:
I am trying to use 
mail("xxx@gmail.com","some message!","test"); 
to send email, it not works. how should i do? should i install some special php-mod-***?

neheb, 2020-06-14:
ping @mhei

mhei, 2020-06-14:
No special php package is required, but did you have a mail binary in your system or did you configured your php.ini accordingly?

NemoAng, 2020-06-14:
> No special php package is required, but did you have a mail binary in your system or did you configured your php.ini accordingly?

Hi, I am not familiar with these. would you mind telling me how to fix it?My configuration as below:
`root@OPEN_WRT_N2:/usr/bin# ls -al|grep php
-rwxr-xr-x    1 root     root             0 Jun 14 09:47 php-cgi
-rwxr-xr-x    1 root     root       3220644 Apr 21 16:05 php-cli
lrwxrwxrwx    1 root     root             7 Apr 28 15:15 php-fcgi -> php-cgi
`
uhttpd configuration:
`	list interpreter '.php=/usr/bin/php-cgi'
`
php.ini:
`[PHP]
date.timezone = "US/Central"

zend.ze1_compatibility_mode = Off

; Language Options

engine = On
;short_open_tag = Off
precision    =  12
y2k_compliance = On
output_buffering = Off
;output_handler =
zlib.output_compression = Off
;zlib.output_compression_level = -1
;zlib.output_handler =
implicit_flush = Off
unserialize_callback_func =
serialize_precision = 100

;open_basedir =
disable_functions =
disable_classes =

; Colors for Syntax Highlighting mode.  Anything that's acceptable in
; <span style="color: ???????"> would work.
;highlight.string  = #DD0000
;highlight.comment = #FF9900
;highlight.keyword = #007700
;highlight.bg      = #FFFFFF
;highlight.default = #0000BB
;highlight.html    = #000000

;ignore_user_abort = On
;realpath_cache_size = 16k
;realpath_cache_ttl = 120

; Miscellaneous

expose_php = On

; Resource Limits

max_execution_time = 30	; Maximum execution time of each script, in seconds.
max_input_time = 60	; Maximum amount of time each script may spend parsing request data.
;max_input_nesting_level = 64
memory_limit = 8M	; Maximum amount of memory a script may consume.

; Error handling and logging

; Error Level Constants:
; E_ALL             - All errors and warnings (includes E_STRICT as of PHP 6.0.0)
; E_ERROR           - fatal run-time errors
; E_RECOVERABLE_ERROR  - almost fatal run-time errors
; E_WARNING         - run-time warnings (non-fatal errors)
; E_PARSE           - compile-time parse errors
; E_NOTICE          - run-time notices (these are warnings which often result
;                     from a bug in your code, but it's possible that it was
;                     intentional (e.g., using an uninitialized variable and
;                     relying on the fact it's automatically initialized to an
;                     empty string)
; E_STRICT			- run-time notices, enable to have PHP suggest changes
;                     to your code which will ensure the best interoperability
;                     and forward compatibility of your code
; E_CORE_ERROR      - fatal errors that occur during PHP's initial startup
; E_CORE_WARNING    - warnings (non-fatal errors) that occur during PHP's
;                     initial startup
; E_COMPILE_ERROR   - fatal compile-time errors
; E_COMPILE_WARNING - compile-time warnings (non-fatal errors)
; E_USER_ERROR      - user-generated error message
; E_USER_WARNING    - user-generated warning message
; E_USER_NOTICE     - user-generated notice message
; E_DEPRECATED      - warn about code that will not work in future versions
;                     of PHP
; E_USER_DEPRECATED - user-generated deprecation warnings
;
; Common Values:
;   E_ALL & ~E_NOTICE  (Show all errors, except for notices and coding standards warnings.)
;   E_ALL & ~E_NOTICE | E_STRICT  (Show all errors, except for notices)
;   E_COMPILE_ERROR|E_RECOVERABLE_ERROR|E_ERROR|E_CORE_ERROR  (Show only errors)
;   E_ALL | E_STRICT  (Show all errors, warnings and notices including coding standards.)
; Default Value: E_ALL & ~E_NOTICE
error_reporting  =  E_ALL & ~E_NOTICE & ~E_STRICT

display_errors = On
display_startup_errors = Off
log_errors = Off
log_errors_max_len = 1024
ignore_repeated_errors = Off
ignore_repeated_source = Off
report_memleaks = On
;report_zend_debug = 0
track_errors = Off
;html_errors = Off
;docref_root = "/phpmanual/"
;docref_ext = .html
;error_prepend_string = "<font color=#ff0000>"
;error_append_string = "</font>"
; Log errors to specified file.
error_log = /var/log/php_errors.log
; Log errors to syslog.
;error_log = syslog

; Data Handling

;arg_separator.output = "&amp;"
;arg_separator.input = ";&"
variables_order = "EGPCS"
request_order = "GP"
register_globals = Off
register_long_arrays = Off
register_argc_argv = On
auto_globals_jit = On
post_max_size = 8M
;magic_quotes_gpc = Off
magic_quotes_runtime = Off
magic_quotes_sybase = Off
auto_prepend_file =
auto_append_file =
default_mimetype = "text/html"
;default_charset = "iso-8859-1"
;always_populate_raw_post_data = On

; Paths and Directories

; UNIX: "/path1:/path2"
;include_path = ".:/php/includes"
;doc_root = "/www"
doc_root = "/overlay/fkccp"
user_dir =
extension_dir = "/usr/lib/php"
enable_dl = On
;cgi.force_redirect = 1
;cgi.nph = 1
;cgi.redirect_status_env = ;
cgi.fix_pathinfo=1
;fastcgi.impersonate = 1;
;fastcgi.logging = 0
;cgi.rfc2616_headers = 0

; File Uploads

file_uploads = On
upload_tmp_dir = "/tmp"
upload_max_filesize = 2M
max_file_uploads = 20

; Fopen wrappers

allow_url_fopen = On
allow_url_include = Off
;from="john@doe.com"
;user_agent="PHP"
default_socket_timeout = 60
;auto_detect_line_endings = Off
`
I have tryed maisend as below:
`mailsend -f user@example.com -smtp smtp.gmail.com -port 465 -t mygmail@gmail.com -sub "testing oneline messages" -cs "us-ascii" -enc-type "7bit" -M "Thi
s is a test"
Socket Error: [11]: Resource temporarily unavailable
Error: Connection is closed unexpectedly
Could not send mail`
I guess my dir505 maybe have enough memory to run it, so i just remove it and hope php mail() can do it.

I use php read/write file correctly. but i can't see any process about php. about php.ini, which options are for mail()? Thanks.

mhei, 2020-06-18:
https://www.php.net/manual/en/mail.configuration.php

closed 2020-06-18 by mhei