openwrt/packages #12496: Qmi wont work well when signal changes

JesseYang2020, 2020-06-13:
There are two cases:
1. When the machine start up, if there is no signal, then qmi wont connect forever
2. When the machien start up, if there has good signal, then qmi will work well. But if it went to the bad signal place for some minutes, it will loss connect forever even you go back to the good signal place.

neheb, 2020-06-14:
ping @aleksander0m

aleksander0m, 2020-06-15:
Yep, I can understand these issues happening.

The netifd MM protocol handler will *try* to bring the connection up. If it cannot, due to any reason, the connection bring up procedure fails. I guess what's happening is that netifd doesn't retry to bring up the connection up automatically. In the openwrt integrations I've done for different clients, I've always added an external process that would monitor the state of the wwan connection to attempt to bring it up if it was down.

@neheb is there any way to ask netifd to keep on attempting to bring up the connection if the initial attempt failed?

JesseYang2020, 2020-06-15:
@aleksander0m @neheb as case one, the uqmi process stuck all the time. If you kill the uqmi process by band, it can reconnect to the internet.