openwrt/packages #12517: samba4-lib: compile error

SuLingGG, 2020-06-14:
Description:

I add  `CONFIG_PACKAGE_libunwind=y` but the error still presence:

```
2020-06-14T06:02:58.6546049Z make[2]: Entering directory '/home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/feeds/packages/net/samba4'
2020-06-14T06:02:58.9485131Z mkdir -p /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/bin/targets/bcm27xx/bcm2711/packages /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/build_dir/target-aarch64_cortex-a72_musl/samba-4.12.3/ipkg-aarch64_cortex-a72/samba4-libs/CONTROL /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-aarch64_cortex-a72_musl/pkginfo
2020-06-14T06:02:58.9511141Z install -d -m0755 /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/build_dir/target-aarch64_cortex-a72_musl/samba-4.12.3/ipkg-aarch64_cortex-a72/samba4-libs/usr/lib
2020-06-14T06:02:58.9535878Z cp -fpR /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/build_dir/target-aarch64_cortex-a72_musl/samba-4.12.3/ipkg-install/usr/lib/*.so* /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/build_dir/target-aarch64_cortex-a72_musl/samba-4.12.3/ipkg-aarch64_cortex-a72/samba4-libs/usr/lib/
2020-06-14T06:02:58.9652266Z # rpath-install
2020-06-14T06:02:58.9669605Z cp -fpR /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/build_dir/target-aarch64_cortex-a72_musl/samba-4.12.3/ipkg-install/usr/lib/samba /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/build_dir/target-aarch64_cortex-a72_musl/samba-4.12.3/ipkg-aarch64_cortex-a72/samba4-libs/usr/lib/
2020-06-14T06:02:58.9902464Z find /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/build_dir/target-aarch64_cortex-a72_musl/samba-4.12.3/ipkg-aarch64_cortex-a72/samba4-libs -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
2020-06-14T06:02:59.3412916Z Package samba4-libs is missing dependencies for the following libraries:
2020-06-14T06:02:59.3419753Z libunwind-aarch64.so.8
2020-06-14T06:02:59.3425719Z Makefile:449: recipe for target '/home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/bin/packages/aarch64_cortex-a72/packages/samba4-libs_4.12.3-1_aarch64_cortex-a72.ipk' failed
2020-06-14T06:02:59.3426433Z make[2]: Leaving directory '/home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/feeds/packages/net/samba4'
2020-06-14T06:02:59.3434644Z make[2]: *** [/home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/bin/packages/aarch64_cortex-a72/packages/samba4-libs_4.12.3-1_aarch64_cortex-a72.ipk] Error 1
2020-06-14T06:02:59.3434839Z time: package/feeds/packages/samba4/compile#0.57#0.28#0.69
2020-06-14T06:02:59.3470882Z make[1]: *** [package/feeds/packages/samba4/compile] Error 2
2020-06-14T06:02:59.3471219Z package/Makefile:111: recipe for target 'package/feeds/packages/samba4/compile' failed
2020-06-14T06:02:59.3471612Z make[1]: Leaving directory '/home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt'
2020-06-14T06:02:59.3481574Z /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/include/toplevel.mk:222: recipe for target 'package/compile' failed
2020-06-14T06:02:59.3488043Z make: *** [package/compile] Error 2
2020-06-14T06:02:59.3530141Z ##[error]Process completed with exit code 2.
```

Andy2244, 2020-06-14:
Yes there are some new "optional" libs samba 4.12.x pulls in, on my local builds and the PR CircleCI it did build without libunwind/ICU. Need to investigate whats going on with there lib handling again.

Andy2244, 2020-06-14:
check #12519

PR #12519 (2020-06-15): samba4: unbundle libunwind, icu
Maintainer: me
Compile tested: arm (master)
Run tested: arm/qemu (master)

Description:
* add unbundle libunwind, icu patches
* fixes #12517

Unbundle optional libs, samba otherwise would pull in if found, but has no direct enable/disable switches in there build system.
closed 2020-06-15 by neheb