openwrt/packages #12518: open-vm-tools: Memory leak

Fail-Safe, 2020-06-14:
Maintainer: @srchack 
Environment: OpenWrt SNAPSHOT, r13552-cd09f26660, x86_64

Description:
There appears to be a memory leak in `vmtoolsd`. Memory consumption creeps up steadily over time and does not release. For example, after 2.5 days of uptime, `vmtoolsd` is consuming >24% memory (out of 512MB), and it continues to increase.

Restarting the service returns memory consumption to the same level as what `vmtoolsd` consumes on my other Linux VMs. Therefore, I believe this bug is unique to the OpenWrt package.

Screenshot:
[image]

```

root@OpenWrt:~# cat /etc/os-release 
NAME="OpenWrt"
VERSION="SNAPSHOT"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt SNAPSHOT"
VERSION_ID="snapshot"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r13552-cd09f26660"
OPENWRT_BOARD="x86/64"
OPENWRT_ARCH="x86_64"
OPENWRT_TAINTS="no-all busybox"
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt SNAPSHOT r13552-cd09f26660"

root@OpenWrt:~# opkg list | grep open-vm-tools
open-vm-tools - 11.1.0-1 - open-vm-tools
```

Fail-Safe, 2020-06-15:
Additional information, in case it's relevant:

```
root@OpenWrt:~# cat /var/log/vmware-vmtoolsd-root.log
[2020-06-14T21:38:51.166Z] [ message] [vmtoolsd] Log caching is enabled with maxCacheEntries=4096.
[2020-06-14T21:38:51.166Z] [ message] [vmtoolsd] Core dump limit set to -1
[2020-06-14T21:38:51.169Z] [ message] [vmtoolsd] Cannot load message catalog for domain 'vmtoolsd', language 'C', catalog dir '/usr/share/open-vm-tools'.
```

```
root@OpenWrt:~# cat /var/log/vmware-vmsvc-root.log
[2020-06-14T21:38:51.169Z] [ message] [vmsvc] Log caching is enabled with maxCacheEntries=4096.
[2020-06-14T21:38:51.170Z] [ message] [vmsvc] Core dump limit set to -1
[2020-06-14T21:38:51.170Z] [ message] [vmtoolsd] Tools Version: 11.1.0.19887 (build-16036546)
[2020-06-14T21:38:51.179Z] [ message] [vmsvc] Cannot load message catalog for domain 'hgfsServer', language 'C', catalog dir '/usr/share/open-vm-tools'.
[2020-06-14T21:38:51.179Z] [ message] [vmtoolsd] Plugin 'hgfsServer' initialized.
[2020-06-14T21:38:51.179Z] [ message] [vmsvc] Cannot load message catalog for domain 'vix', language 'C', catalog dir '/usr/share/open-vm-tools'.
[2020-06-14T21:38:51.179Z] [ message] [vmtoolsd] Plugin 'vix' initialized.
[2020-06-14T21:38:51.179Z] [ message] [vmsvc] Cannot load message catalog for domain 'guestInfo', language 'C', catalog dir '/usr/share/open-vm-tools'.
[2020-06-14T21:38:51.179Z] [ message] [vmtoolsd] Plugin 'guestInfo' initialized.
[2020-06-14T21:38:51.179Z] [ message] [vmsvc] Cannot load message catalog for domain 'powerops', language 'C', catalog dir '/usr/share/open-vm-tools'.
[2020-06-14T21:38:51.179Z] [ message] [vmtoolsd] Plugin 'powerops' initialized.
[2020-06-14T21:38:51.179Z] [ message] [vmsvc] Cannot load message catalog for domain 'timeSync', language 'C', catalog dir '/usr/share/open-vm-tools'.
[2020-06-14T21:38:51.179Z] [ message] [vmtoolsd] Plugin 'timeSync' initialized.
[2020-06-14T21:38:51.179Z] [ message] [vmsvc] Cannot load message catalog for domain 'vmbackup', language 'C', catalog dir '/usr/share/open-vm-tools'.
[2020-06-14T21:38:51.179Z] [ message] [vmtoolsd] Plugin 'vmbackup' initialized.
[2020-06-14T21:38:51.183Z] [ message] [vix] VixTools_ProcessVixCommand: command 62
[2020-06-14T21:38:51.186Z] [ message] [vix] ToolsDaemonTcloReceiveVixCommand: command 62, additionalError = 22
```

kagurazakakotori, 2020-06-19:
Can you tell me the version of your esxi/workstation. I can't make the same memory leak happen on my platform.

Fail-Safe, 2020-06-19:
I'm running ESXi 6.7.0 Update 3 (Build 16075168). I do see that Build 16316930 is available, I could try to upgrade to it and see if it makes a difference.

Are you also running ESXi 6.7?

Here's another screen capture at 4.5+ days:

[image]

kagurazakakotori, 2020-06-20:
Yesterday I made a clean install of the latest openwrt snapshot `r13597-be56b29707`, and open-vm-tools 11.1.0-1 from [downloads.openwrt.org](https://downloads.openwrt.org/), the memory leak doesn't happen. Maybe this is a problem caused by openwrt itself. BTW, I am running ESXi 6.7.0 Update 3 (Build 16316930).

[image]

The link below tells the difference between `r13552-cd09f26660` and `r13597-be56b29707`.
https://github.com/openwrt/openwrt/compare/cd09f26...be56b29

Fail-Safe, 2020-06-21:
I updated to Build 16316930 on my ESXi host a couple hours ago. Does not appear to have resolved the issue I’m seeing with vmtoolsd memory usage. I’m going to keep an eye on it for a while longer and see if I can find any other causes. It’s very perplexing to me.

kagurazakakotori, 2020-06-21:
Maybe you should try updating your openwrt version. In my opinion, the leak might be caused by a broken memory allocator. Snapshots builds may contain bugs.

validev, 2021-08-06:
I see this issue, too, on 19.07-SNAPSHOT, r11260+14 / r11354+22 / r11328+12 / open-vm-tools 10.3.10-1 with ESXi 5.1.0 ( I know it's stinking old, but never the less)

Please fix.

closed 2025-05-24 by Fail-Safe