openwrt/packages #12520: ksmbd modprobe not working

lucize, 2020-06-14:
Maintainer: @Andy2244 
Environment: latest

Description:
modprobe ksmbd seems not to work (at least on my builds) but insmod works
maybe this in the init script:
`[ ! -e /sys/module/ksmbd ] && insmod /lib/modules/$(uname -r)/ksmbd.ko 2> /dev/null`
instead of
`[ ! -e /sys/module/ksmbd ] && modprobe ksmbd 2> /dev/null`

Andy2244, 2020-06-14:
This is on master or 19.07 and can you check what modinfo shows? Kinda strange that modprobe is not working.

lucize, 2020-06-14:
fantastic, now after editing and reverting the line seems to work on reboot but a service restart gives
```
daemon.err ksmbd: modprobe of ksmbd module failed, can\'t start ksmbd!
 modprobe -v ksmbd
1 module could not be probed
- ksmbd

```

dl12345, 2020-06-17:
And if you do a `depmod -a` before the modprobe? I noticed some similar strange module loading behaviour on a very recent build and the `depmod -a` fixed it, but I haven't had the time to track down the cause yet

Andy2244, 2020-06-25:
I did just update/test ksmbd 3.2.2 and cant reproduce this problem. Start/Restart and changing options works fine for me.

lucize, 2020-07-05:
I have other modules that are not in main feed failing so I don't know what is the real problem

closed 2020-07-05 by lucize

stefanct, 2025-11-09:
I seem to have exactly the same problem in the current HEAD of 24.10.4. can someone please re-open this isse and lend me a hand in debugging it? it seems really similar as in modprobe cannot find the module by its name:

    # modprobe -v ksmbd
    modprobe: FATAL: Module ksmbd not found in directory /lib/modules/6.6.110

neither does `modinfo ksmbd`:

   modinfo: ERROR: Module ksmbd not found.

however, the module is there and seems fine in general:

    # modinfo /lib/modules/6.6.110/ksmbd.ko
    filename:       /lib/modules/6.6.110/ksmbd.ko
    softdep:        pre: crc32
    softdep:        pre: gcm
    softdep:        pre: ccm
    softdep:        pre: aead2
    softdep:        pre: sha512
    softdep:        pre: sha256
    softdep:        pre: cmac
    softdep:        pre: aes
    softdep:        pre: nls
    softdep:        pre: md5
    softdep:        pre: hmac
    softdep:        pre: ecb
    license:        GPL
    depends:        oid_registry,nls_ucs2_utils,cifs_arc4,asn1_decoder
    intree:         Y
    name:           ksmbd
    vermagic:       6.6.110 SMP mod_unload aarch64

in my case i am additionally running into the problem that depmod segfaults... :)

the last lines of `strace depmod -a`:

    newfstatat(AT_FDCWD, "/lib/modules/6.6.110/btrfs.ko", {st_mode=S_IFREG|0644, st_size=1547264, ...}, 0) = 0
    mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f9887f000
    munmap(0x7f9887f000, 4096)              = 0
    --- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_MAPERR, si_addr=0x7f9887f045} ---

stefanct, 2025-11-09:
Well, I think i figured it out with the help of dwfreed on IRC. Rebuilding the image.

the culprit was that i uninformedly selected `CONFIG_PACKAGE_kmod=y` because I thought that's the right thing if i want to manipulate kernel modules at runtime... apparently that's not the way and one should stick to the tools that come with ubox.