openwrt/packages #12522: Openwrt plans to support <TAB> command auto completion?

q158073378252010, 2020-06-15:
After ssh login, use the <TAB> key to automatically complete the command? Other linux distributions seem to use the project [bash-completion](https://github.com/scop/bash-completion) Openwrt has plans to support?
 Is it possible to port the project bash-completion to openwrt?

neheb, 2020-06-16:
By default, OpenWrt uses busybox ash, not bash.

q158073378252010, 2020-06-16:
openwrt can install **bash** as default

neheb, 2020-06-16:
I don't really know how much demand there is for this.

OTOB, fish shell is available right now with completion.

dl12345, 2020-06-17:
@q158073378252010 bash is already available for openwrt and supports tab completion. Just install the `bash` package then edit `/etc/passwd` and change the shell to `/bin/bash`

q158073378252010, 2020-06-17:
@dl12345 
I already do this. Doesn't seem to work.
Found the package "bash-completion" from other Linux distributions.
Openwrt_x86 platform is easy to solve. ( make && make install )
smip, arm needs "Makefile". This is a bit difficult for me, so let me ask.

dl12345, 2020-06-17:
I use it and it works fine for me

acooler15, 2021-09-11:
> @dl12345
> I already do this. Doesn't seem to work.
> Found the package "bash-completion" from other Linux distributions.
> Openwrt_x86 platform is easy to solve. ( make && make install )
> smip, arm needs "Makefile". This is a bit difficult for me, so let me ask.

```
#
# Copyright (C) 2007-2015 OpenWrt.org
#
# This is free software, licensed under the GNU General Public License v2.
# See /COPYING for more information.
#

include $(TOPDIR)/rules.mk

PKG_NAME:=bash-completion
PKG_VERSION:=2.11
PKG_RELEASE:=1

PKG_SOURCE:=$(PKG_NAME)-$(PKG_VERSION).tar.xz
PKG_SOURCE_URL:=https://github.com/scop/bash-completion/releases/download/$(PKG_VERSION)/
PKG_HASH:=73a8894bad94dee83ab468fa09f628daffd567e8bef1a24277f1e9a0daf911ac
PKG_MAINTAINER:=acooler15 <acooler15@foxmail.com>
PKG_LICENSE:=GPL-2.0-or-later
PKG_LICENSE_FILES:=COPYING

PKG_INSTALL:=1
PKG_BUILD_PARALLEL:=1

include $(INCLUDE_DIR)/package.mk

define Package/bash-completion/conffiles
/etc/bash_completion.d/
endef

define Package/bash-completion
	SECTION:=utils
	CATEGORY:=Utilities
	SUBMENU:=Shells
	TITLE:=Programmable completion for the Bash shell
	DEPENDS:=+bash +procps-ng-ps
	URL:=https://github.com/scop/bash-completion
endef

define Package/bash-completion/description
	bash-completion is a collection of command line command completions for the Bash shell, 
	collection of helper functions to assist in creating new completions, 
	and set of facilities for loading completions automatically on demand, as well as installing them.
endef

define Package/bash-completion/install
	$(INSTALL_DIR) $(1)/etc/profile.d
	$(CP) $(PKG_INSTALL_DIR)/etc/profile.d/* $(1)/etc/profile.d/
	$(INSTALL_DIR) $(1)/usr/share/bash-completion
	$(CP) $(PKG_INSTALL_DIR)/usr/share/bash-completion/* $(1)/usr/share/bash-completion/
	$(INSTALL_DIR) $(1)/etc/bash_completion.d
endef

$(eval $(call BuildPackage,bash-completion))

```

q158073378252010, 2021-09-12:
> 
> 
> > @dl12345 
> >  我已经这样做了。 似乎不起作用。 
> >  从其他 Linux 发行版找到包“bash-completion”。 
> >  openwrt_x86平台很容易解决。 ( make && make install ) 
> >  smip，arm 需要“Makefile”。 这对我来说有点困难，所以让我问一下。
> 
> ```
> #
> # Copyright (C) 2007-2015 OpenWrt.org
> #
> # This is free software, licensed under the GNU General Public License v2.
> # See /COPYING for more information.
> #
> 
> include $(TOPDIR)/rules.mk
> 
> PKG_NAME:=bash-completion
> PKG_VERSION:=2.11
> PKG_RELEASE:=1
> 
> PKG_SOURCE:=$(PKG_NAME)-$(PKG_VERSION).tar.xz
> PKG_SOURCE_URL:=https://github.com/scop/bash-completion/releases/download/$(PKG_VERSION)/
> PKG_HASH:=73a8894bad94dee83ab468fa09f628daffd567e8bef1a24277f1e9a0daf911ac
> PKG_MAINTAINER:=acooler15 <acooler15@foxmail.com>
> PKG_LICENSE:=GPL-2.0-or-later
> PKG_LICENSE_FILES:=COPYING
> 
> PKG_INSTALL:=1
> PKG_BUILD_PARALLEL:=1
> 
> include $(INCLUDE_DIR)/package.mk
> 
> define Package/bash-completion/conffiles
> /etc/bash_completion.d/
> endef
> 
> define Package/bash-completion
> 	SECTION:=utils
> 	CATEGORY:=Utilities
> 	SUBMENU:=Shells
> 	TITLE:=Programmable completion for the Bash shell
> 	DEPENDS:=+bash +procps-ng-ps
> 	URL:=https://github.com/scop/bash-completion
> endef
> 
> define Package/bash-completion/description
> 	bash-completion is a collection of command line command completions for the Bash shell, 
> 	collection of helper functions to assist in creating new completions, 
> 	and set of facilities for loading completions automatically on demand, as well as installing them.
> endef
> 
> define Package/bash-completion/install
> 	$(INSTALL_DIR) $(1)/etc/profile.d
> 	$(CP) $(PKG_INSTALL_DIR)/etc/profile.d/* $(1)/etc/profile.d/
> 	$(INSTALL_DIR) $(1)/usr/share/bash-completion
> 	$(CP) $(PKG_INSTALL_DIR)/usr/share/bash-completion/* $(1)/usr/share/bash-completion/
> 	$(INSTALL_DIR) $(1)/etc/bash_completion.d
> endef
> 
> $(eval $(call BuildPackage,bash-completion))
> ```

closed 2021-09-12 by q158073378252010