openwrt/packages #12523: keepalived: support master/backup configurations (enhancement)

pgaufillet, 2020-06-15:
Maintainer: @scrpi @feckert
Environment: ath79, Openwrt r13405-02f08056bc

Description: when switching a router between MASTER and BACKUP keepalived modes, it is frequent to have to change its configuration. For example when becoming MASTER, DHCP has to be started, VPN has to be opened, routes and firewall rules have to be adapted. An elegant solution to this need would be to offer folders `master` and `backup` in `/etc/keepalived`, hosting each specific UCI configuration files. These can then be linked from `/etc/config` depending on keepalived mode thanks to a script also in charge of restarting the corresponding services.

feckert, 2020-06-16:
I think this functionality is already presented.
You could use the following keepalived user script https://github.com/openwrt/packages/blob/master/net/keepalived/files/keepalived.user This is called (if presented) on every state change. This file is treated as a shell script. So I you could add any command you want to this script.

pgaufillet, 2020-07-11:
Of course. My point is that it could go a step further, proposing by default master/backup openwrt configuration files that are swapped when the operational mode of keepalived changes.

feckert, 2020-07-21:
That's not the job of the keepalived.
If we add that to the package, then the package inflates.
Most people don't need that and also have to customize their use case.
Also, it's not generic and very individual what happens in a switch.

The best thing to do is to make a new package and put keepalived as dependency and put your scripts in /etc/hotplug.d/keepalived.

closed 2020-07-21 by feckert