openwrt/packages #12529: Stubby: Upgrade to latest upstream

jawadcogilent, 2020-06-16:
Maintainer: @jonathanunderwood
Environment: arm_cortex-a9_vfpv3, Linksys WRT1900AC, 19.07.3 (put here arch, model, OpenWrt version)

Description:

```
Kindly upgrade to latest upstream version of stubby 0.3.0
```

jonathanunderwood, 2020-06-17:
There are a few issues with the latest getdns and stubby releases, especially around openssl, that are causing me to hold off for the next release of getdns before updating. Please see the projects issue tracker for more detail.

jonathanunderwood, 2020-07-01:
getdns update in progress in #12679  - testing welcome

jonathanunderwood, 2020-07-01:
stubby update in progress #12682 - testing welcome

jonathanunderwood, 2020-07-06:
stubby 0.3.0 and getdns 1.6.0 have been merged into master and 19.07 branches.

**This can be closed**

jawadcogilent, 2020-07-07:
Many thanks.

closed 2020-07-07 by jawadcogilent

bogi788, 2020-07-07:
the update seems to have brought https://github.com/openwrt/packages/issues/7019 back

jonathanunderwood, 2020-07-07:
> the update seems to have brought #7019 back

Please open a new issue and provide more details including what version of OpenWRT you're using, the output you see etc.