openwrt/packages #12532: samba4: Samba package not building due to missing python dependency

thagabe, 2020-06-17:
Maintainer: @Andy2244 
Environment: cortex9, ARM, mevbu, WRT3200ACM

Description:

Build halts at "missing dependency 'python-devel' which is not present in the build environment. I am building on master branch and pulling all commits. Last week I build an img just fine before the changes on this commit (1bc2f4f3c6216ece87f0821d68e4e25f821dc382). 

````
I'll add the logs here.
````

thagabe, 2020-06-17:
My mistake, looks like these new commit introduced new python3-distutils dependency for the build environment.

Andy2244, 2020-06-17:
mhh i do build python3/host explicitly and this should also be used by samba4, instead of the existing host tools. I also verified the build against this docker [environment](https://github.com/Andy2244/docker-openwrt-sdk-base/blob/master/Dockerfile), which did build fine. So there should not be any additional dependency on the host?

DazzyWalkman, 2020-06-19:
Same here. I built using glibc.
Due to https://github.com/openwrt/openwrt/commit/b933f9cf0cb254e368027cad6d5799e45b237df5, 
a `make dirclean` was performed before `make`.
`--disable-python --without-ad-dc` in Makefile does not work.

`sudo apt install python3-distutils` on my ubuntu 20.04 building host solved this problem. 

related output:
```
['/opt/openwrt/staging_dir/host/bin/python3', '-c', "from distutils.sysconfig import get_config_var, get_python_lib\n\nprint(repr(get_python_lib(standard_lib=0, prefix='/opt/openwrt/staging_dir/hostpkg') or ''))"]
err: Traceback (most recent call last):
  File "<string>", line 1, in <module>
ModuleNotFoundError: No module named 'distutils.sysconfig'

from /opt/openwrt/build_dir/hostpkg/samba-4.12.3: The distutils module is unusable: install "python-devel"?
```

Andy2244, 2020-06-19:
> `--disable-python --without-ad-dc` in Makefile does not work.

What do you mean by "does not work"?
Keep in mind those settings just disable python/ad-dc support on the target, the waf buildsystem still needs python to actually cross-build.

> `sudo apt install python3-distutils` on my ubuntu 20.04 building host solved this problem.

It might be the case that debian 9 does install python3-distutils alongside its python3 meta package and thats why i don't need this in staging. Will try to fix this thanks.

jefferyto, 2020-06-19:
Looks like configure found build system Python (`/opt/openwrt/staging_dir/host/bin/python3`) instead of host Python (python3/host, in `staging_dir/hostpkg`).

Andy2244, 2020-06-19:
> Looks like configure found build system Python (`/opt/openwrt/staging_dir/host/bin/python3`) instead of host Python (python3/host, in `staging_dir/hostpkg`).

I think i found the problem, on the samba4/host build step i do not explicitly set the python3 bin location, so i assume samba finds the org. host/bin instead of hostpkg/bin. Will recheck what paths it uses on my build env and fix this issue.

Andy2244, 2020-06-19:
@DazzyWalkman @thagabe can you test if #12555 fixes the issues? Aka uninstall python3-distutils beforehand.

thagabe, 2020-06-20:
Works without host pkg!

PR #12555 (2020-06-19): samba4: fix python3 host paths
Maintainer: me
Compile tested: arm (master)
Run tested: arm/qemu (master)

Description:
* fix python3 host paths, ensure we use build hostpkg tools
* fixes #12532 ? _(need feedback)_
closed 2020-06-20 by thagabe