openwrt/packages #12538: Help php5 openwrt 18.04 eror builf

predators46, 2020-06-18:
`   /home/open/cortex53/build_dir/target-aarch64_cortex-a53_musl/php-5.6.32/ext/intl/intl_convertcpp.h:28:25: error: 'UnicodeString' was not declared in this scope
 int intl_stringFromChar(UnicodeString &ret, char *str, int32_t str_len, UErrorCode *status);
                         ^~~~~~~~~~~~~
/home/open/cortex53/build_dir/target-aarch64_cortex-a53_musl/php-5.6.32/ext/intl/intl_convertcpp.h:28:25: note: suggested alternative:
In file included from /home/open/cortex53/build_dir/target-aarch64_cortex-a53_musl/php-5.6.32/ext/intl/intl_convertcpp.h:26:0,
                 from /home/open/cortex53/build_dir/target-aarch64_cortex-a53_musl/php-5.6.32/ext/intl/intl_convertcpp.cpp:21:
/home/open/cortex53/staging_dir/target-aarch64_cortex-a53_musl/usr/include/unicode/unistr.h:286:20: note:   'icu_61::UnicodeString'
 class U_COMMON_API UnicodeString : public Replaceable `

neheb, 2020-06-18:
ping @mhei

mhei, 2020-06-18:
Please post the output of `scripts/diffconfig.sh` so that I can try to re-produce your issue.

predators46, 2020-06-18:
> Please post the output of `scripts/diffconfig.sh` so that I can try to re-produce your issue.

`   #!/bin/sh
grep \^CONFIG_TARGET_ .config | head -n3 > tmp/.diffconfig.head
grep \^CONFIG_TARGET_DEVICE_ .config >> tmp/.diffconfig.head
grep '^CONFIG_ALL=y' .config >> tmp/.diffconfig.head
grep '^CONFIG_ALL_KMODS=y' .config >> tmp/.diffconfig.head
grep '^CONFIG_ALL_NONSHARED=y' .config >> tmp/.diffconfig.head
grep '^CONFIG_DEVEL=y' .config >> tmp/.diffconfig.head
grep '^CONFIG_TOOLCHAINOPTS=y' .config >> tmp/.diffconfig.head
grep '^CONFIG_BUSYBOX_CUSTOM=y' .config >> tmp/.diffconfig.head
grep '^CONFIG_TARGET_PER_DEVICE_ROOTFS=y' .config >> tmp/.diffconfig.head
./scripts/config/conf --defconfig=tmp/.diffconfig.head -w tmp/.diffconfig.stage1 Config.in >/dev/null
./scripts/kconfig.pl '>+' tmp/.diffconfig.stage1 .config >> tmp/.diffconfig.head
./scripts/config/conf --defconfig=tmp/.diffconfig.head -w tmp/.diffconfig.stage2 Config.in >/dev/null
./scripts/kconfig.pl '>' tmp/.diffconfig.stage2 .config >> tmp/.diffconfig.head
cat tmp/.diffconfig.head
rm -f tmp/.diffconfig tmp/.diffconfig.head
 `

predators46, 2020-06-18:
I use [imagebuilder](https://downloads.openwrt.org/releases/18.06.4/targets/sunxi/cortexa53/openwrt-sdk-18.06.4-sunxi-cortexa53_gcc-7.3.0_musl.Linux-x86_64.tar.xz)

mhei, 2020-06-19:
Sorry, I don't need the script itself - I can checkout myself - you must run the script and post the output. But since you say that you're using the OpenWrt SDK, please give me a _detailed_ instruction, about what you're doing and what you want to achieve. Please take my point of view: when I want to re-produce your problem I must be able to exactly follow your steps, so do not assume any knowledge on my side and show me all steps.

closed 2020-06-19 by predators46