openwrt/packages #12554: lame-lib: missing libmvec.so.1 with glibc

DazzyWalkman, 2020-06-19:
Package: lame-lib 3.100

PKG_MAINTAINER: @thess 

I was compiling  x86_64 snapshot r13592 with glibc toolchain and lame-lib was selected as a dependent package by ffmpeg, starting from `make dirclean` 

The compilation failed with error: Missing libmvec.so.1

I manually copied libmvec.so* from 
`./staging_dir/target-x86_64_glibc/usr/lib/` 
to 
`./build_dir/target-x86_64_glibc/lame-3.100/ipkg-x86_64/lame-lib/usr/lib/ `

Then the package compilation could be resumed without error.

plntyk, 2020-06-19:
apparently libmvec is a glibc component - introduced in 2.22 , its a packaging issue of glibc 
see https://sourceware.org/glibc/wiki/libmvec

also from buildroot fix:
"On some architectures (namely x86-64), glibc may provide a libmvec
library since glibc 2.22, which programs built with gcc OpenMP support
might get linked to."

potential other OpenMP packets might be affected

plntyk, 2020-06-19:
cannot replicate that error with ffmpeg in trunk - please post the output of scripts/diffconfig.sh

DazzyWalkman, 2020-06-19:
Here is the output of diffconfig. Thanks.
[diffconfig.out.zip](https://github.com/openwrt/packages/files/4805558/diffconfig.out.zip)

DazzyWalkman, 2020-09-28:
Disabling LAME-LIB_OPTIMIZE_SPEED silence the problem.  It seems that "-O3 -ffast-math" will pull in libmvec.so.