openwrt/packages #12557: pythom3-cffi: packages/cffi/cffi_opcode.py': No such file or directory

peterwillcn, 2020-06-20:
pythom3-cffi

```
t/master/openwrt/staging_dir/target-aarch64_generic_musl" _python_prefix="/usr" _python_exec_prefix="/usr" ; fi; )
    Running setup.py install for cffi: finished with status 'error'
    Running setup.py install for cffi: finished with status 'error'
    ERROR: Command errored out with exit status 1:
    copying cffi/error.py -> build/lib.linux2-3.8/cffi
    copying cffi/_cffi_errors.h -> build/lib.linux2-3.8/cffi
    c/_cffi_backend.c:6245:5: warning: 'ffi_prep_closure' is deprecated: use ffi_prep_closure_loc instead [-Wdeprecated-declarations]
    error: could not delete '/home/openwrt/master/openwrt/staging_dir/hostpkg/lib/python3.8/site-packages/cffi/cffi_opcode.py': No such file or directory
    ERROR: Command errored out with exit status 1:
    copying cffi/error.py -> build/lib.linux2-3.8/cffi
    copying cffi/_cffi_errors.h -> build/lib.linux2-3.8/cffi
    c/_cffi_backend.c:6245:5: warning: 'ffi_prep_closure' is deprecated: use ffi_prep_closure_loc instead [-Wdeprecated-declarations]
    copying build/lib.linux2-3.8/cffi/error.py -> /home/openwrt/master/openwrt/staging_dir/hostpkg/lib/python3.8/site-packages/cffi
    copying build/lib.linux2-3.8/cffi/_cffi_errors.h -> /home/openwrt/master/openwrt/staging_dir/hostpkg/lib/python3.8/site-packages/cffi
    error: [Errno 2] No such file or directory
ERROR: Command errored out with exit status 1: /home/openwrt/master/openwrt/staging_dir/hostpkg/bin/python3.8 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/home/openwrt/master/openwrt/tmp/pip-install-6kx8xxye/cffi/setup.py'"'"'; __file__='"'"'/home/openwrt/master/openwrt/tmp/pip-install-6kx8xxye/cffi/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"
'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /home/openwrt/master/openwrt/tmp/pip-record-5z2gmb5m/install-record.txt --single-version-externally-managed --compile Check the logs for full command output.
ERROR: Command errored out with exit status 1: /home/openwrt/master/openwrt/staging_dir/hostpkg/bin/python3.8 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/home/openwrt/master/openwrt/tmp/pip-install-gcpj775p/cffi/setup.py'"'"'; __file__='"'"'/home/openwrt/master/openwrt/tmp/pip-install-gcpj775p/cffi/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"
'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /home/openwrt/master/openwrt/tmp/pip-record-p_8opfw5/install-record.txt --single-version-externally-managed --compile Check the logs for full command output.
make[3]: *** [/home/openwrt/master/openwrt/build_dir/target-aarch64_generic_musl/pypi/bcrypt-3.1.7/.built] Error 1
make[2]: *** [package/feeds/lang/python/bcrypt/compile] Error 2
make[3]: *** [/home/openwrt/master/openwrt/build_dir/target-aarch64_generic_musl/pypi/cryptography-2.9.2/.built] Error 1
make[2]: *** [package/feeds/lang/python/python-cryptography/compile] Error 2
```

peterwillcn, 2020-06-20:
@jefferyto

jefferyto, 2020-06-20:
The error is occurring during host install for cffi, not the python3-cffi package.

I see references to both bcrypt and python-cryptography; these two packages do not depend on each other. (They both depend on host cffi though, there could be some conflict there.) What are you trying to build?

jefferyto, 2020-06-24:
Were you building with more than one job at a time (`-j` in the make command line)?

BKPepe, 2020-07-16:
Can this be closed?

jefferyto, 2020-07-16:
I think so but it would be best if @peterwillcn can confirm.

closed 2020-09-19 by neheb