openwrt/packages #12565: MiniDLNA still runs automatically by default when not enabled!!

zxlhhyccc, 2020-06-21:
Even if enable again stop also cannot stop!!
[image]

neheb, 2020-06-22:
which OpenWrt version?

I recommend switching to Gerbera.

zxlhhyccc, 2020-06-22:
> which OpenWrt version?
> 
> I recommend switching to Gerbera.

master branch。。。

neheb, 2020-06-22:
Try latest miniDLNA package.

zxlhhyccc, 2020-06-23:
> Try latest miniDLNA package.

The e problem of automatic operation with the new software package remains unresolved.

antonlacon, 2020-06-28:
It's an issue with the luci app. I did a fresh install today and verified luci says it's enabled and serving 0 files. However, minidlna doesn't appear in the process list, in top, or have /var/run/minidlna created (meaning it had started and created a db file) - IOW, minidlna wasn't running.