openwrt/packages #12568: gcc -O2 flag breaks under ccache

Grommish, 2020-06-21:
https://github.com/openwrt/packages/blob/f2639763768f7981ae7971697e0b7dce1100b557/net/squid/Makefile#L73

When building with ccache, the CXX_FLAGS -O2 flag is invalid for ccache, causing the compile to fail.

neheb, 2020-07-08:
Not quite. The problem is that OpenWrt's ccache support uses two CMake variables, which can break assumptions. See https://github.com/openwrt/openwrt/blob/master/include/cmake.mk#L20

I fixed an example of that here: https://github.com/openwrt/packages/blob/master/libs/libpng/patches/200-ccache.patch

karlp, 2020-07-09:
So... cmake is inferior to make? ;)

Grommish, 2020-07-09:
After talking with @jow- I decided ti just scrap using ccache completely.  In his words, it was a whole lot of pain for negligible gain, but it's good for anyone else who might see this.. Thanks @neheb!

closed 2020-07-09 by Grommish