openwrt/packages #12578: ddns-scripts:ddns doesnt recpect the setting for check interval

rezad1393, 2020-06-23:
ddns-scripts

Maintainer: @\<github-user> (find it by checking history of the package Makefile)

Linux openwrt 4.14.180 #0 SMP armv7l GNU/Linux / linsys1200wrt/ 19.07.3

Description:
using ddns with duckdns (which I dont think is important).
it check for ip updates every some minutes like this

logread -f |grep ddns        
Jun 23 10:49:57 openwrt ddns-scripts[15840]: duckdns: PID '15840' started at 2020-06-23 16:19
Jun 23 10:49:57 openwrt ddns-scripts[15840]: duckdns: Starting main loop at 2020-06-23 16:19
Jun 23 10:50:01 openwrt ddns-scripts[15840]: duckdns: Update successful - IP '-------' send
Jun 23 10:51:18 openwrt ddns-scripts[2585]: duckdns: Updating IP at DDNS provider failed - starting retry 143/0
Jun 23 10:51:21 openwrt ddns-scripts[2585]: duckdns: Update successful - IP '----------' send
Jun 23 10:54:01 openwrt ddns-scripts[25610]: duckdns: Updating IP at DDNS provider failed - starting retry 36/0
Jun 23 10:54:03 openwrt ddns-scripts[25610]: duckdns: Update successful - IP '--------' send
(IP hidden for privacy)
```
and this is the config
config ddns 'global'
        option ddns_dateformat '%F %R'
        option ddns_loglines '250'
        option upd_privateip '0'

config service 'duckdns'
        option password 'xxxxx'
        option use_https '1'
        option lookup_host 'xxxx.duckdns.org'
        option domain 'xxxx.duckdns.org'
        option enabled '1'
        option service_name 'duckdns.org'
        option use_syslog '1'
        option ip_source 'web'
        option interface 'wan'
        option ip_network 'wan'
        option check_interval '1'
        option check_unit 'hours'
        option ip_url 'https://checkip.amazonaws.com/'
        option cacert '/etc/ssl/certs/ca-certificates.crt'

rezad1393, 2020-06-23:
this is more from the log after restarting the service:
```

logread |grep ddns

Jun 23 11:09:25 openwrt ddns-scripts[15840]: duckdns: PID '15840' terminated by 'SIGTERM' at 2020-06-23 16:39
Jun 23 11:09:27 openwrt ddns-scripts[17167]: duckdns: PID '17167' started at 2020-06-23 16:39
Jun 23 11:09:27 openwrt ddns-scripts[17167]: duckdns: Starting main loop at 2020-06-23 16:39
Jun 23 11:09:29 openwrt ddns-scripts[17167]: duckdns: Update successful - IP 'xxxxxxxxx' send
Jun 23 11:11:32 openwrt ddns-scripts[2585]: duckdns: Updating IP at DDNS provider failed - starting retry 147/0
Jun 23 11:11:35 openwrt ddns-scripts[2585]: duckdns: Update successful - IP 'xxxxxxxxx' send
Jun 23 11:14:06 openwrt ddns-scripts[25610]: duckdns: Updating IP at DDNS provider failed - starting retry 38/0
Jun 23 11:14:09 openwrt ddns-scripts[25610]: duckdns: Update successful - IP 'xxxxxxxxx' send
Jun 23 11:16:35 openwrt ddns-scripts[2585]: duckdns: Updating IP at DDNS provider failed - starting retry 148/0
Jun 23 11:16:38 openwrt ddns-scripts[2585]: duckdns: Update successful - IP 'xxxxxxxxx' send
Jun 23 11:21:38 openwrt ddns-scripts[2585]: duckdns: Updating IP at DDNS provider failed - starting retry 149/0
Jun 23 11:21:41 openwrt ddns-scripts[2585]: duckdns: Update successful - IP 'xxxxxxxxx' send
Jun 23 11:24:09 openwrt ddns-scripts[25610]: duckdns: Updating IP at DDNS provider failed - starting retry 39/0
Jun 23 11:24:12 openwrt ddns-scripts[25610]: duckdns: Update successful - IP 'xxxxxxxxx' send
Jun 23 11:26:41 openwrt ddns-scripts[2585]: duckdns: Updating IP at DDNS provider failed - starting retry 150/0
Jun 23 11:26:44 openwrt ddns-scripts[2585]: duckdns: Update successful - IP 'xxxxxxxxx' send
```

jaimet, 2020-07-02:
Just to corroborate this report (I'm actually in the middle of reporting another bug in ddns-scripts, but I noticed this "check-interval" problem too, so I thought that I would add this information):

```
root@router:~# /usr/lib/ddns/dynamic_dns_updater.sh -v 2 -- start
 133128       : ************ ************** ************** **************
 133128  note : PID '15980' started at 2020-07-02 13:31
 133128       : ddns version  : 2.7.8-21
 133128       : uci configuration:
ddns.myddns_ipv4.cacert='/etc/ssl/certs'
ddns.myddns_ipv4.check_interval='60'
ddns.myddns_ipv4.check_unit='seconds'
ddns.myddns_ipv4.domain='mydomain.nsupdate.info'
ddns.myddns_ipv4.enabled='1'
ddns.myddns_ipv4.ip_network='wan'
ddns.myddns_ipv4.ip_source='network'
ddns.myddns_ipv4.lookup_host='mydomain.nsupdate.info'
ddns.myddns_ipv4.password='mypassword'
ddns.myddns_ipv4.service_name='nsupdate.info'
ddns.myddns_ipv4.use_https='1'
ddns.myddns_ipv4.use_syslog='1'
ddns.myddns_ipv4.username='mydomain.nsupdate.info'
ddns.myddns_ipv4=service
 133128       : verbose mode  : 2 - run once, NO retry on error
 133128       : check interval: 300 seconds
 133128       : force interval: 259200 seconds
 133128       : retry interval: 60 seconds
 133129       : retry counter : 0 times
 133129       : No old process
 133129       : last update: 2020-07-01 23:46
 133129       : Detect registered/public IP
 133129       : #> /usr/bin/nslookup mydomain.nsupdate.info  >/var/run/ddns/myddns_ipv4.dat 2>/var/run/ddns/myddns_ipv4.err
 133129       : Registered IP '5.6.7.9' detected
 133129  info : Starting main loop at 2020-07-02 13:31
 133129       : Detect local IP on 'network'
 133129       : Local IP '5.6.7.8' detected on network 'wan'
 133129       : Update needed - L: '5.6.7.8' <> R: '5.6.7.9'
 133130  WARN : No valid certificate file '/etc/ssl/certs' for HTTPS communication - TERMINATE
 133130  WARN : PID '15980' exit WITH ERROR '1' at 2020-07-02 13:31
```
My /etc/config/ddns contains:

```
  option check_interval '60'
  option check_unit 'seconds'

```
and yet the above contains the following 2 lines:
```
ddns.myddns_ipv4.check_interval='60'
133128       : check interval: 300 seconds

```

_Note to self_ 1: Look through [dynamic_dns_functions.sh](https://github.com/openwrt/packages/blob/master/net/ddns-scripts/files/dynamic_dns_functions.sh) to see if you can work out what's going wrong.

_Note to self_ 2: when &lt;local&gt;=&lt;registered&gt;, dynamic_dns_updater.sh skips the update but then waits the check interval. Huh? What's the logic *supposed* to be? Does [the current ddns-gargoyle source](https://github.com/ericpaulbishop/gargoyle/tree/master/package/ddns-gargoyle) help? (ebishop wrote them both - see [here](https://www.gargoyle-router.com/phpbb/viewtopic.php?t=24
) and [here](https://www.gargoyle-router.com/phpbb/viewtopic.php?p=16280#p16280
))

brianjmurrell, 2023-02-23:
> My /etc/config/ddns contains:
> 
> ```
>   option check_interval '60'
>   option check_unit 'seconds'
> ```

[Minimum is 5 minutes](https://github.com/openwrt/packages/blob/4f07ac43e21ee3794310590d5fbecc4035f83796/net/ddns-scripts/files/usr/lib/ddns/dynamic_dns_updater.sh#L292).

But frankly I disagree with that since the check is being done at the authoritative nameserver.  Not all "services" have such horrible update turn-around.  Using `nsupdate` with a Bind9 for example, the update is immediate and can be checked immediately.  #20564 and #20565

tim-connect, 2025-06-18:
Confirming that this still an issue, even with the minimum 5 minutes interval:

```
...
ddns.cloudflare_ddns_mail_depstone.check_interval='5'
ddns.cloudflare_ddns_mail_depstone.domain='****hidden****'
ddns.cloudflare_ddns_mail_depstone.enabled='1'
ddns.cloudflare_ddns_mail_depstone.interface='wan'
ddns.cloudflare_ddns_mail_depstone.ip_network='wan'
ddns.cloudflare_ddns_mail_depstone.ip_source='network'
ddns.cloudflare_ddns_mail_depstone.lookup_host='****hidden****'
ddns.cloudflare_ddns_mail_depstone.password='***PW***'
ddns.cloudflare_ddns_mail_depstone.service_name='cloudflare.com-v4'
ddns.cloudflare_ddns_mail_depstone.use_https='1'
ddns.cloudflare_ddns_mail_depstone.username='Bearer'
ddns.cloudflare_ddns_mail_depstone=service
 115855       : verbose mode  : 0 - run normal, NO console output
 115855       : check interval: 300 seconds

...

```

Check interval of 5 minutes (even with minutes explicitly stated, despite minutes being the default unit), has no effect. 600 seconds every time.