openwrt/packages #12617: python-attrs: compilation error

feckert, 2020-06-26:
Maintainer: @jefferyto 
Environment: x86_64, APU3, OpenWrt master

Description:

I get the following error if I enable docker-compose package.
This error does not happen all the time on every build.
I thought it has to do with parallel builds, so I disable this in the Makefile for the package with `PKG_BUILD_PARALLEL:=0`.
But that didn't help either.
@jefferyto Any hints for this issue would be create!

```
Traceback (most recent call last):
  File "setup.py", line 105, in <module>
    setup(
  File "/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/staging_dir/hostpkg/lib/python3.8/site-packages/setuptools/__init__.py", line 145, in setup
    return distutils.core.setup(**attrs)
  File "/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/staging_dir/target-x86_64_musl/usr/lib/python3.8/distutils/core.py", line 148, in setup
    dist.run_commands()
  File "/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/staging_dir/target-x86_64_musl/usr/lib/python3.8/distutils/dist.py", line 966, in run_commands
    self.run_command(cmd)
  File "/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/staging_dir/target-x86_64_musl/usr/lib/python3.8/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/staging_dir/hostpkg/lib/python3.8/site-packages/setuptools/command/install.py", line 61, in run
    return orig.install.run(self)
  File "/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/staging_dir/target-x86_64_musl/usr/lib/python3.8/distutils/command/install.py", line 545, in run
    self.run_command('build')
  File "/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/staging_dir/target-x86_64_musl/usr/lib/python3.8/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/staging_dir/target-x86_64_musl/usr/lib/python3.8/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/staging_dir/target-x86_64_musl/usr/lib/python3.8/distutils/command/build.py", line 135, in run
    self.run_command(cmd_name)
  File "/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/staging_dir/target-x86_64_musl/usr/lib/python3.8/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/staging_dir/target-x86_64_musl/usr/lib/python3.8/distutils/dist.py", line 983, in run_command
    cmd_obj = self.get_command_obj(command)
  File "/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/staging_dir/target-x86_64_musl/usr/lib/python3.8/distutils/dist.py", line 857, in get_command_obj
    klass = self.get_command_class(command)
  File "/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/staging_dir/hostpkg/lib/python3.8/site-packages/setuptools/dist.py", line 840, in get_command_class
    self.cmdclass[command] = cmdclass = ep.load()
  File "/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/staging_dir/hostpkg/lib/python3.8/site-packages/pkg_resources/__init__.py", line 2443, in load
    return self.resolve()
  File "/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/staging_dir/hostpkg/lib/python3.8/site-packages/pkg_resources/__init__.py", line 2449, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/staging_dir/hostpkg/lib/python3.8/site-packages/setuptools/command/build_py.py", line 15, in <module>
    from setuptools.lib2to3_ex import Mixin2to3
  File "/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/staging_dir/hostpkg/lib/python3.8/site-packages/setuptools/lib2to3_ex.py", line 12, in <module>
    from lib2to3.refactor import RefactoringTool, get_fixers_from_package
  File "/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/staging_dir/target-x86_64_musl/usr/lib/python3.8/lib2to3/refactor.py", line 26, in <module>
    from .fixer_util import find_root
  File "/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/staging_dir/target-x86_64_musl/usr/lib/python3.8/lib2to3/fixer_util.py", line 7, in <module>
    from .pygram import python_symbols as syms
  File "/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/staging_dir/target-x86_64_musl/usr/lib/python3.8/lib2to3/pygram.py", line 32, in <module>
    python_grammar = driver.load_packaged_grammar("lib2to3", _GRAMMAR_FILE)
  File "/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/staging_dir/target-x86_64_musl/usr/lib/python3.8/lib2to3/pgen2/driver.py", line 155, in load_packaged_grammar
    return load_grammar(grammar_source)
  File "/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/staging_dir/target-x86_64_musl/usr/lib/python3.8/lib2to3/pgen2/driver.py", line 130, in load_grammar
    g.load(gp)
File "/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/staging_dir/target-x86_64_musl/usr/lib/python3.8/lib2to3/pgen2/grammar.py", line 95, in load
    d = pickle.load(f)
EOFError: Ran out of input
Makefile:43: recipe for target '/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/build_dir/target-x86_64_musl/pypi/attrs-19.3.0/.built' failed
```

jefferyto, 2020-06-26:
I haven't been able to reproduce this error, I'm not entirely sure how I can reproduce this. If there is any more information you can add, that would be great.

feckert, 2020-06-29:
@jefferyto thanks for your reply. As already said the error does not occur with every **fresh** openwrt build! I think this is related to the docker-compose. Next time it comes up I'll upload the log.

feckert, 2020-12-01:
It seems to be fixed. I haven't had any build errors for a long time

feckert, 2021-02-18:
@jefferyto I have to reopen this issue unfortunately. I have now repeatedly problem with this error. I can not always reproduce it. Sometimes the build runs through sometimes not. And it is not always the same package that does not build. The last error was with python-urllib3.

```
feckert@bbworker01:/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt$ cat logs/package/feeds/packages/python-urllib3/
compile.txt   download.txt
feckert@bbworker01:/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt$ cat logs/package/feeds/packages/python-urllib3/compile.txt
running install
running build
running build_py
Traceback (most recent call last):
  File "/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/build_dir/target-x86_64_musl/pypi/urllib3-1.25.11/setup.py", line 39, in <module>
    setup(
  File "/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/staging_dir/hostpkg/lib/python3.9/site-packages/setuptools/__init__.py", line 165, in setup
    return distutils.core.setup(**attrs)
  File "/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/staging_dir/target-x86_64_musl/usr/lib/python3.9/distutils/core.py", line 148, in setup
    dist.run_commands()
  File "/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/staging_dir/target-x86_64_musl/usr/lib/python3.9/distutils/dist.py", line 966, in run_commands
    self.run_command(cmd)
  File "/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/staging_dir/target-x86_64_musl/usr/lib/python3.9/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/staging_dir/hostpkg/lib/python3.9/site-packages/setuptools/command/install.py", line 61, in run
    return orig.install.run(self)
  File "/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/staging_dir/target-x86_64_musl/usr/lib/python3.9/distutils/command/install.py", line 546, in run
    self.run_command('build')
  File "/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/staging_dir/target-x86_64_musl/usr/lib/python3.9/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/staging_dir/target-x86_64_musl/usr/lib/python3.9/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/staging_dir/target-x86_64_musl/usr/lib/python3.9/distutils/command/build.py", line 135, in run
    self.run_command(cmd_name)
  File "/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/staging_dir/target-x86_64_musl/usr/lib/python3.9/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/staging_dir/target-x86_64_musl/usr/lib/python3.9/distutils/dist.py", line 983, in run_command
    cmd_obj = self.get_command_obj(command)
  File "/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/staging_dir/target-x86_64_musl/usr/lib/python3.9/distutils/dist.py", line 857, in get_command_obj
    klass = self.get_command_class(command)
  File "/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/staging_dir/hostpkg/lib/python3.9/site-packages/setuptools/dist.py", line 768, in get_command_class
    self.cmdclass[command] = cmdclass = ep.load()
  File "/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/staging_dir/hostpkg/lib/python3.9/site-packages/pkg_resources/__init__.py", line 2461, in load
    return self.resolve()
  File "/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/staging_dir/hostpkg/lib/python3.9/site-packages/pkg_resources/__init__.py", line 2467, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/staging_dir/hostpkg/lib/python3.9/site-packages/setuptools/command/build_py.py", line 16, in <module>
    from setuptools.lib2to3_ex import Mixin2to3
  File "/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/staging_dir/hostpkg/lib/python3.9/site-packages/setuptools/lib2to3_ex.py", line 13, in <module>
    from lib2to3.refactor import RefactoringTool, get_fixers_from_package
  File "/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/staging_dir/target-x86_64_musl/usr/lib/python3.9/lib2to3/refactor.py", line 26, in <module>
    from .fixer_util import find_root
  File "/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/staging_dir/target-x86_64_musl/usr/lib/python3.9/lib2to3/fixer_util.py", line 7, in <module>
    from .pygram import python_symbols as syms
  File "/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/staging_dir/target-x86_64_musl/usr/lib/python3.9/lib2to3/pygram.py", line 32, in <module>
    python_grammar = driver.load_packaged_grammar("lib2to3", _GRAMMAR_FILE)
  File "/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/staging_dir/target-x86_64_musl/usr/lib/python3.9/lib2to3/pgen2/driver.py", line 155, in load_packaged_g
    return load_grammar(grammar_source)
  File "/home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/staging_dir/target-x86_64_musl/usr/lib/python3.9/lib2to3/pgen2/driver.py", line 130, in load_grammar
    g.load(gp)
[... cut ...]
    d = pickle.load(f)
EOFError: Ran out of input
make[4]: *** [Makefile:45: /home/bbworker/bbworker/owrt_master_x86_64/System6/build/openwrt/build_dir/target-x86_64_musl/pypi/urllib3-1.25.11/.built] Error 1
time: package/feeds/packages/python-urllib3/compile#1.00#0.03#1.19
```

I can't figure out the error message. Google could not help me there either. Only so much that apparently the file is empty he wants to use, or that it has changed. Could this be a problem with parallel python package builds?

jefferyto, 2021-06-13:
I'm getting an error compiling cryptodev-linux when trying to compile Python with the x86-64 sdk so I haven't been able to replicate this.

Looking at the stack trace, it's weird how host setuptools (in `staging_dir/hostpkg/lib/python3.9/`) loads distutils and lib2to3 from `staging_dir/target-x86_64_musl/usr/lib/python3.9/`. Perhaps this is the cause of the error?

@commodo any ideas?

commodo, 2021-06-14:
> I'm getting an error compiling cryptodev-linux when trying to compile Python with the x86-64 sdk so I haven't been able to replicate this.
>
> Looking at the stack trace, it's weird how host setuptools [...]

no idea yet
will try to reproduce on my end

feckert, 2021-09-07:
I don't know why, but I haven't seen the issue for a few weeks now. Therefore I will close this here

closed 2021-09-07 by feckert