openwrt/packages #12621: golang: GO_PKG_LDFLAGS_X broken

aparcar, 2020-06-26:
Maintainer: @jefferyto 
Environment: x86/64 and mvebu/cortexa9

Description:

The commit https://github.com/openwrt/packages/commit/25a7f0045e078cbd91d2d1de71ddd617ec4ff165 breaks the usage of GO_PKG_LDFLAGS_X so that the variables set in the package Makefile are no longer used. This causes trouble at least for the `syncthing` package as other clients won't sync with a `unknown-dev` version.

I reverted the patch and the version for `syncthing` works again:

```
# before
syncthing unknown-dev "Fermium Flea" (go1.14.4 linux-arm) unknown@unknown 1970-01-01 00:00:00 UTC

#reverted
syncthing v1.4.0 "Fermium Flea" (go1.14.4 linux-arm) openwrt@openwrt 2020-06-25 21:09:05 UTC
```

jefferyto, 2020-06-26:
`GO_PKG_LDFLAGS_X` isn't working for syncthing now because the variable is defined after golang-package.mk is included. My implicit assumption is that `GO_PKG_*` variables should be defined before golang-package.mk is included (like how `PKG_*` variables should be defined before package.mk is included).

I will update golang-package.mk to allow `GO_PKG_*` to be defined after the include, but I still think they should be defined before unless there is a reason not to.

aparcar, 2020-06-27:
Please keep it as is and I'll update the Makefile of syncthing. I think it's better to be strict with the order to avoid future corner cases.

closed 2020-06-27 by aparcar

jefferyto, 2020-06-27:
I think there may be some situations when `GO_PKG_*` variables need to be defined/redefined after the include (although I can't think of any at the moment), so I still think making it more flexible is a good thing.

I appreciate you updating syncthing (#12624) as well :smile: