openwrt/packages #12625: master branch miniupnpd missing “libcap”！！！

zxlhhyccc, 2020-06-27:
Please make sure that the issue subject starts with `<package-name>: `

Also make sure that the package is maintained in this repository and not in base which should be submitted at https://bugs.openwrt.org or in the LuCI repository which should be submitted at https://github.com/openwrt/luci/issues.

Issues related to releases below 18.06 and forks are not supported or maintained and will be closed.

# Issue template (remove lines from top till here)

Maintainer: @\<github-user> (find it by checking history of the package Makefile)
Environment: (put here arch, model, OpenWrt version)

Description:

```
Formating code blocks by wrapping them with pairs of ```
```

neheb, 2020-06-27:
Reproduced. Will whip up a patch shortly.

neheb, 2020-06-28:
Fixed this.

closed 2020-06-28 by neheb