openwrt/packages #12626: domoticz: no custom menu visable and no plugin pages

mabachel, 2020-06-27:
Maintainer: @ammubhave  @neheb @dwmw2 @stintel
Environment: (ipq806x, Netgear R7800, OpenWrt master)

Description:
I am running a recent master build of OpenWRT on my Netgear R7800 and compiled and installed node 12 from here: https://github.com/nxhack/openwrt-node-packages

I also installed [zigbee2mqtt](https://github.com/koenkk/zigbee2mqtt) via `npm ci` and python3, mosquitto-ssl and domoticz. Additionally I copied [domoticz-plugins-manager](https://github.com/stas-demydiuk/domoticz-plugins-manager) and [domoticz-zigbee2mqtt-plugin](https://github.com/stas-demydiuk/domoticz-zigbee2mqtt-plugin) to `/etc/domoticz/plugins`.

From my understanding and from what can bee seen on the screenshots of the plugins github pages ther sould be as _Custom_ menu in domoticz and a page where I can interact with the domoticz plugins. But as you can see in the attached  screenshots I don't see anything like that.
[image]
[image]

However from the syslog I can see that a _plugin custom page_ is being installed
```
daemon.info domoticz: Domoticz is starting up....
daemon.info domoticz: Domoticz running...
daemon.info domoticz: Active notification Subsystems: fcm, http (2/13)
daemon.info domoticz: Starting shared server on: :::6144
daemon.info domoticz: (Mosquitto) Installing plugin custom page...
daemon.info domoticz: (Mosquitto) Installing plugin custom page completed.
daemon.info domoticz: (Mosquitto) Connected to MQTT Server: 127.0.0.1:1883
daemon.info domoticz: (Mosquitto) Zigbee2mqtt bridge is online
daemon.info domoticz: (Mosquitto) Joining new devices is disabled on the zigbee bridge
daemon.info domoticz: (Mosquitto) Received available devices list from bridge
daemon.info domoticz: (Mosquitto) Received groups list from bridge
```

How can I interact with my Plugins? Are they supported at all? What python3 (full, light, or base) packages are really necessary?

mabachel, 2020-06-27:
One more question:
In the config it says:
```
# CAUTION - by default, /var is not persistent accross reboots
# Don't forget the trailing / - domoticz requires it
option userdata '/var/lib/domoticz/'
```
What is the suggested solution to get a persistent configuration of domoticz?

mabachel, 2020-06-27:
I also added The Plugins on the Hardware Page:

[image]

and I can this under devices:
[image]

neheb, 2020-09-20:
domoticz is no longer maintained. You're on your own I'm afraid.

dwmw2, 2020-09-20:
I am taking over. Apologies for the delay.  I'm not familiar with the plugins and have avoided perl but the file system layout should work. I have been using a setup with symlinks to an external file system but will investigate.