openwrt/packages #12629: ctop: Buildbot errors

jefferyto, 2020-06-27:
Maintainer: @jmarcet 
Environment: multiple

Description:

Buildbots have been failing to build ctop:

* https://downloads.openwrt.org/snapshots/faillogs/aarch64_cortex-a53/packages/ctop/compile.txt
* https://downloads.openwrt.org/snapshots/faillogs/arm_cortex-a15_neon-vfpv4/packages/ctop/compile.txt
* https://downloads.openwrt.org/snapshots/faillogs/i386_pentium4/packages/ctop/compile.txt

Quoting from the error log for aarch64_cortex-a53:

```
install: cannot stat '/builder/shared-workdir/build/sdk/build_dir/target-aarch64_cortex-a53_musl/ctop-0.7.3/.go_work/build/bin/ctop': No such file or directory
Makefile:52: recipe for target '/builder/shared-workdir/build/sdk/build_dir/target-aarch64_cortex-a53_musl/ctop-0.7.3/.pkgdir/ctop.installed' failed
make[3]: *** [/builder/shared-workdir/build/sdk/build_dir/target-aarch64_cortex-a53_musl/ctop-0.7.3/.pkgdir/ctop.installed] Error 1
```

It's unfortunate that the CircleCI check uses ath79, because it is mips the package (with `@!mips`) was probably not actually compiled in #11159 (I didn't check the CircleCI log). If the check was arm or i386 (or any platform other than x86_64) it would have failed.

jmarcet, 2020-07-13:
How should we got about this then, with `DEPENDS:=$(GO_ARCH_DEPENDS) @(aarch64||arm||x86_64)` like [docker-ce](https://github.com/openwrt/packages/blob/e2fe9bda3d6a2755f36031d8d69c6ea8e08ba715/utils/docker-ce/Makefile#L46) ?

I don't have time right now to try building for different architectures, I'm the middle of a relocation to another country.

jefferyto, 2020-07-14:
aarch64 and arm are exactly the architectures where build failures are occurring.

If you are busy and don't have the time for this, just say so. If you want to fix this please at least read the issue description clearly.

I don't want to be mean, but a basic difference between creating a package for yourself and creating/maintaining a package for others is testing on architectures used by others.

jmarcet, 2020-07-14:
From the logs I see all the components seem to be built, but the final executable is not found where it should be. Without testing locally I do not know what the actual error is.

About my current lack of time. I'm not relocating all the time, it will be a matter of two weeks until I am settled in my destination.

Also, I never stated it had been tested on anything else other than x86_64 when I did the PR.

jefferyto, 2020-07-15:
When Go is cross-compiling, it places the binary in a subdirectory named for the target architecture (don't ask me why).

There are other (more direct) ways of getting the path, but the rough consensus seems to be to use:

```make
$(call GoPackage/Package/Install/Bin,$(PKG_INSTALL_DIR))
```

then copy the binary from `$(PKG_INSTALL_DIR)` (e.g. [dnscrypt-proxy2](https://github.com/openwrt/packages/blob/335efed8cb56bcc2c140efcb42a3b70fb4b4a14c/net/dnscrypt-proxy2/Makefile#L42), [libnetwork](https://github.com/openwrt/packages/blob/335efed8cb56bcc2c140efcb42a3b70fb4b4a14c/utils/libnetwork/Makefile#L41)). This will probably fix the build issue but please test it locally first.

Also, I see the Makefile sets a number of `MAKE_*` variables. Since it is using the default Go packages build recipe (from golang-package.mk), these variables have no effect.

I also suggest setting [`GO_PKG_LDFLAGS_X`](https://github.com/openwrt/packages/blob/335efed8cb56bcc2c140efcb42a3b70fb4b4a14c/lang/golang/golang-package.mk#L100-L105) for [`main.version` and `main.build`](https://github.com/bcicen/ctop/blob/v0.7.3/Makefile#L5).

The default ctop build also includes [`-tags release`](https://github.com/bcicen/ctop/blob/4741b276e4bbaa41a67d62443239d50b5a936623/Makefile#L12). You can override `Build/Compile` like [syncthing](https://github.com/openwrt/packages/blob/335efed8cb56bcc2c140efcb42a3b70fb4b4a14c/utils/syncthing/Makefile#L45-L47) to set this, but FYI I think I will add a variable soon (something like `GO_PKG_BUILD_TAGS`) that will handle these build tags.

jefferyto, 2020-07-16:
You can use `GO_PKG_TAGS` to set the `release` build tag (#12833).

jefferyto, 2020-08-15:
ping @jmarcet

jefferyto, 2020-10-13:
The build error was fixed in #13479. Maintainer does not appear to be interested in making improvements based on my [other suggestions](https://github.com/openwrt/packages/issues/12629#issuecomment-658670356) (`MAKE_*` vars, `-X` flags, `release` build tag). Closing.

closed 2020-10-13 by jefferyto