openwrt/packages #12673: python3-pip: Compile Error

mabachel, 2020-07-01:
Maintainer: @commodo @jefferyto 
Environment: (ipq806x, Netgear R7800, OpenWrt master)

Description:
I tried compiling python3 today, but it ended up with the floowing error. It worked just fine a few days ago.
```
make[4]: Entering directory '/home/relentless/Playground/cc2531-home-automation-stack/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/Python-3.8.3'
Rebuilding with profile guided optimizations:
rm -f profile-clean-stamp
make build_all CFLAGS_NODIST=" -fprofile-use -fprofile-correction" LDFLAGS_NODIST=""
make[5]: Entering directory '/home/relentless/Playground/cc2531-home-automation-stack/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/Python-3.8.3'
 CC='arm-openwrt-linux-muslgnueabi-gcc' LDSHARED='arm-openwrt-linux-muslgnueabi-gcc -shared -L/home/relentless/Playground/cc2531-home-automation-stack/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/usr/lib -L/home/relentless/Playground/cc2531-home-automation-stack/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/lib -L/home/relentless/Playground/cc2531-home-automation-stack/openwrt/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-8.4.0_musl_eabi/usr/lib -L/home/relentless/Playground/cc2531-home-automation-stack/openwrt/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-8.4.0_musl_eabi/lib -znow -zrelro -L/home/relentless/Playground/cc2531-home-automation-stack/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/Python-3.8.3 -L/home/relentless/Playground/cc2531-home-automation-stack/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/usr/lib -L/home/relentless/Playground/cc2531-home-automation-stack/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/lib -L/home/relentless/Playground/cc2531-home-automation-stack/openwrt/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-8.4.0_musl_eabi/usr/lib -L/home/relentless/Playground/cc2531-home-automation-stack/openwrt/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-8.4.0_musl_eabi/lib -znow -zrelro -L/home/relentless/Playground/cc2531-home-automation-stack/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/Python-3.8.3 -flto -fuse-linker-plugin -ffat-lto-objects -flto-partition=none -g ' OPT='-DNDEBUG -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -ffile-prefix-map=/home/relentless/Playground/cc2531-home-automation-stack/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/Python-3.8.3=Python-3.8.3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro' 	_TCLTK_INCLUDES='' _TCLTK_LIBS='' 	_PYTHON_PROJECT_BASE=/home/relentless/Playground/cc2531-home-automation-stack/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/Python-3.8.3 _PYTHON_HOST_PLATFORM=linux2 PYTHONPATH=/home/relentless/Playground/cc2531-home-automation-stack/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/Python-3.8.3/Lib:/home/relentless/Playground/cc2531-home-automation-stack/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/Python-3.8.3/build/lib.linux2-3.8 _PYTHON_SYSCONFIGDATA_NAME=_sysconfigdata /home/relentless/Playground/cc2531-home-automation-stack/openwrt/staging_dir/hostpkg/bin/python3.8 ./setup.py  build
running build
running build_ext
INFO: Can't locate Tcl/Tk libs and/or headers

Python build finished successfully!
The necessary bits to build these optional modules were not found:
_tkinter              _uuid                 nis                
To find the necessary bits, look in setup.py in detect_modules() for the module's name.

The following modules found by detect_modules() in setup.py, have been
built by the Makefile instead, as configured by the Setup files:
_abc                  atexit                pwd                
time                  zlib                                     

running build_scripts
copying /home/relentless/Playground/cc2531-home-automation-stack/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/Python-3.8.3/Tools/scripts/pydoc3 -> build/scripts-3.8
copying /home/relentless/Playground/cc2531-home-automation-stack/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/Python-3.8.3/Tools/scripts/idle3 -> build/scripts-3.8
copying /home/relentless/Playground/cc2531-home-automation-stack/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/Python-3.8.3/Tools/scripts/2to3 -> build/scripts-3.8
renaming build/scripts-3.8/pydoc3 to build/scripts-3.8/pydoc3.8
renaming build/scripts-3.8/idle3 to build/scripts-3.8/idle3.8
renaming build/scripts-3.8/2to3 to build/scripts-3.8/2to3-3.8
make[5]: Leaving directory '/home/relentless/Playground/cc2531-home-automation-stack/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/Python-3.8.3'
make[4]: Leaving directory '/home/relentless/Playground/cc2531-home-automation-stack/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/Python-3.8.3'
# Use host pip to install python-setuptools
/home/relentless/Playground/cc2531-home-automation-stack/openwrt/staging_dir/hostpkg/bin/pip3.8 --disable-pip-version-check --cache-dir "/home/relentless/Playground/cc2531-home-automation-stack/openwrt/dl/pip-cache" install --ignore-installed --root=/home/relentless/Playground/cc2531-home-automation-stack/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/Python-3.8.3/install-setuptools --prefix=/usr /home/relentless/Playground/cc2531-home-automation-stack/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/Python-3.8.3/Lib/ensurepip/_bundled/setuptools-41.2.0-py2.py3-none-any.whl
bash: /home/relentless/Playground/cc2531-home-automation-stack/openwrt/staging_dir/hostpkg/bin/pip3.8: /home/relentless/Playground/openwrt/staging_dir/hostpkg/bin/python3.8: bad interpreter: No such file or directory
make[3]: *** [Makefile:361: /home/relentless/Playground/cc2531-home-automation-stack/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/Python-3.8.3/.built] Error 126
make[3]: Leaving directory '/home/relentless/Playground/cc2531-home-automation-stack/openwrt/feeds/packages/lang/python/python3'
time: package/feeds/packages/python3/compile#0.22#0.09#0.31
make[2]: *** [package/Makefile:113: package/feeds/packages/python3/compile] Error 2
make[2]: Leaving directory '/home/relentless/Playground/cc2531-home-automation-stack/openwrt'
make[1]: *** [package/Makefile:107: /home/relentless/Playground/cc2531-home-automation-stack/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/relentless/Playground/cc2531-home-automation-stack/openwrt'
make: *** [/home/relentless/Playground/cc2531-home-automation-stack/openwrt/include/toplevel.mk:224: world] Error 2

```

jefferyto, 2020-07-01:
Please try doing a clean checkout and build.

mabachel, 2020-07-01:
Thanks for the fast reply but I still see errors:
```
[relentless@machine openwrt]$ git clean -f
[relentless@machine openwrt]$ git pull
Already up to date.
[relentless@machine openwrt]$ git checkout master
M	feeds.conf.default
Already on 'master'
Your branch is up to date with 'origin/master'.
[relentless@machine openwrt]$ scripts/feeds update -a
Updating feed 'packages' from 'https://git.openwrt.org/feed/packages.git' ...
Already up to date.
Create index file './feeds/packages.index' 
Updating feed 'luci' from 'https://git.openwrt.org/project/luci.git' ...
Already up to date.
Create index file './feeds/luci.index' 
Updating feed 'routing' from 'https://git.openwrt.org/feed/routing.git' ...
Already up to date.
Create index file './feeds/routing.index' 
Updating feed 'telephony' from 'https://git.openwrt.org/feed/telephony.git' ...
Already up to date.
Create index file './feeds/telephony.index' 
Updating feed 'freifunk' from 'https://github.com/freifunk/openwrt-packages.git' ...
Already up to date.
Create index file './feeds/freifunk.index' 
Updating feed 'node' from 'https://github.com/nxhack/openwrt-node-packages.git' ...
Already up to date.
Create index file './feeds/node.index' 
[relentless@machine openwrt]$ scripts/feeds install -a
Installing all packages from feed packages.
Installing all packages from feed luci.
Installing all packages from feed routing.
Installing all packages from feed telephony.
Installing all packages from feed freifunk.
Installing all packages from feed node.
[relentless@machine openwrt]$ make -j1 V=s

STRIPPED

make[4]: Entering directory '/home/relentless/Playground/cc2531-home-automation-stack/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/Python-3.8.3'
Rebuilding with profile guided optimizations:
rm -f profile-clean-stamp
make build_all CFLAGS_NODIST=" -fprofile-use -fprofile-correction" LDFLAGS_NODIST=""
make[5]: Entering directory '/home/relentless/Playground/cc2531-home-automation-stack/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/Python-3.8.3'
 CC='arm-openwrt-linux-muslgnueabi-gcc' LDSHARED='arm-openwrt-linux-muslgnueabi-gcc -shared -L/home/relentless/Playground/cc2531-home-automation-stack/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/usr/lib -L/home/relentless/Playground/cc2531-home-automation-stack/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/lib -L/home/relentless/Playground/cc2531-home-automation-stack/openwrt/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-8.4.0_musl_eabi/usr/lib -L/home/relentless/Playground/cc2531-home-automation-stack/openwrt/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-8.4.0_musl_eabi/lib -znow -zrelro -L/home/relentless/Playground/cc2531-home-automation-stack/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/Python-3.8.3 -L/home/relentless/Playground/cc2531-home-automation-stack/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/usr/lib -L/home/relentless/Playground/cc2531-home-automation-stack/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/lib -L/home/relentless/Playground/cc2531-home-automation-stack/openwrt/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-8.4.0_musl_eabi/usr/lib -L/home/relentless/Playground/cc2531-home-automation-stack/openwrt/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-8.4.0_musl_eabi/lib -znow -zrelro -L/home/relentless/Playground/cc2531-home-automation-stack/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/Python-3.8.3 -flto -fuse-linker-plugin -ffat-lto-objects -flto-partition=none -g ' OPT='-DNDEBUG -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -ffile-prefix-map=/home/relentless/Playground/cc2531-home-automation-stack/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/Python-3.8.3=Python-3.8.3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro' 	_TCLTK_INCLUDES='' _TCLTK_LIBS='' 	_PYTHON_PROJECT_BASE=/home/relentless/Playground/cc2531-home-automation-stack/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/Python-3.8.3 _PYTHON_HOST_PLATFORM=linux2 PYTHONPATH=/home/relentless/Playground/cc2531-home-automation-stack/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/Python-3.8.3/Lib:/home/relentless/Playground/cc2531-home-automation-stack/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/Python-3.8.3/build/lib.linux2-3.8 _PYTHON_SYSCONFIGDATA_NAME=_sysconfigdata /home/relentless/Playground/cc2531-home-automation-stack/openwrt/staging_dir/hostpkg/bin/python3.8 ./setup.py  build
running build
running build_ext
INFO: Can't locate Tcl/Tk libs and/or headers

Python build finished successfully!
The necessary bits to build these optional modules were not found:
_tkinter              _uuid                 nis                
To find the necessary bits, look in setup.py in detect_modules() for the module's name.

The following modules found by detect_modules() in setup.py, have been
built by the Makefile instead, as configured by the Setup files:
_abc                  atexit                pwd                
time                  zlib                                     

running build_scripts
copying /home/relentless/Playground/cc2531-home-automation-stack/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/Python-3.8.3/Tools/scripts/pydoc3 -> build/scripts-3.8
[... cut ...]
src-git routing https://git.openwrt.org/feed/routing.git
src-git telephony https://git.openwrt.org/feed/telephony.git
src-git freifunk https://github.com/freifunk/openwrt-packages.git
#src-git video https://github.com/openwrt/video.git
#src-git targets https://github.com/openwrt/targets.git
#src-git management https://github.com/openwrt-management/packages.git
#src-git oldpackages http://git.openwrt.org/packages.git
#src-link custom /usr/src/openwrt/custom-feed
src-git node https://github.com/nxhack/openwrt-node-packages.git
```

jefferyto, 2020-07-01:
I've compiled python3 using both the ipq806x sdk and OpenWrt master and I haven't encountered any errors. I suggest you do a fresh checkout instead of cleaning your existing repo and try again.

If you still cannot compile the package, please include more information about your build system.

BKPepe, 2020-07-01:
I met the same issue on openSUSE Tumbleweed a few days ago while building packages for OpenWrt 19.07 and master. On which operating system are you trying to cross-compile? I think I solved it by installing/updating python3-setuptools.

mabachel, 2020-07-02:
I am on Fedora 32, but I fixed it cloning the repo and tart from scratch. But this time I added python3-setuptools and python3-pip as modules via make menuconfig. I don't know if this is important or not.

Either way.. probably not an issue anymore.. can be closed..?!

jefferyto, 2020-07-07:
I doubt selecting/building python3-setuptools and python3-pip made a difference. I think what @BKPepe meant was installing/updating the openSUSE python3-setuptools package (or the equivalent package for your build system).

I think this can be closed - please reopen if the issue reappears.

closed 2020-07-07 by BKPepe