openwrt/packages #12680: [19.07] Some Python packages are not built due to ValueError: not enough values to unpack

BKPepe, 2020-07-01:
Maintainers: @jefferyto , @commodo 
Env: all
After update of Python3, some packages like zipp, cryptography, dateutil fails with the following error:

```
  File "/builder/shared-workdir/build/sdk/staging_dir/hostpkg/lib/python3.7/site-packages/pip/_vendor/packaging/tags.py", line 642, in _linux_platforms
    _, arch = linux.split("_", 1)
ValueError: not enough values to unpack (expected 2, got 1)
Makefile:35: recipe for target '/builder/shared-workdir/build/sdk/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/pypi/python-zipp-python3/zipp-0.6.0/.built' failed
```

Faillog: https://downloads.openwrt.org/releases/faillogs-19.07/arm_cortex-a9_vfpv3-d16/packages/python-zipp/python3/compile.txt

BKPepe, 2020-07-01:
I think that this could be related to this issue https://github.com/pypa/setuptools/issues/1934

BKPepe, 2020-07-01:
I downgraded setuptools in `staging_dir/hostpkg/lib/python3.7/site-packages` and installed `setuptools-scm` by this command: `build/staging_dir/hostpkg/bin/pip3.7 install setuptools-scm` and then run `make package/python-zipp/{cleancompile}` it worked.

jefferyto, 2020-07-02:
I think the actual issue is pypa/packaging#259. I'm looking into why we are setting `_PYTHON_HOST_PLATFORM=linux2` now, if we can do what Alpine is doing or if we should patch pip's copy of packaging to bypass this.

commodo, 2020-07-02:
> I think the actual issue is [pypa/packaging#259](https://github.com/pypa/packaging/issues/259). I'm looking into why we are setting `_PYTHON_HOST_PLATFORM=linux2` now, if we can do what Alpine is doing or if we should patch pip's copy of packaging to bypass this.

That setting is pretty old.
I think I just copied it from the old Python ~2.7.3 without giving it much thought.
I think it may have been put there by the first person who packaged Python for OpenWrt (before 2015). In the “oldpackages” repo, if that still exists anywhere these days.

jefferyto, 2020-07-02:
Well...

* Host pip got the same environment variables as host Python in 4d94943818e918676a161a218b0778ad14c30471 and 4d0a44d31ad1a387e8b810283960b524e1136b2c. (Not sure how the exact same change can be merged twice without any conflicts...)
* Host Python got `_PYTHON_HOST_PLATFORM=linux2` when I added python-host.mk in b12d7b6db12b0830f021df5058b347f030feac70 (this was synced to python3-host.mk in 3af20278869ad43ead7361616887685231a2fb97). I believe I copied `_PYTHON_HOST_PLATFORM=linux2` from python-package.mk.
* `_PYTHON_HOST_PLATFORM="linux-$(ARCH)"` was changed to `_PYTHON_HOST_PLATFORM=linux2` in python-package.mk in f5f8d83dcdb31a2e0d0948c576ca7e744b09520c (synced to python3-package.mk in 8c057f39d80b609739028c8141a36b7674efa468).
* `_PYTHON_HOST_PLATFORM="linux-$(ARCH)"` was added to python-package.mk in 0141057a59d86cccc7550d0ccbb6f42ed12b41a0 (added to python3-package.mk in 056e565ab646f991ad7b3a9dd95f63ad50248f3b).

Two things:
* I think `_PYTHON_HOST_PLATFORM` should not be set for host pip (and perhaps even host Python), since it is installing/running code for host
* For building packages for target, I'm still trying to understand if/what value of `_PYTHON_HOST_PLATFORM` is appropriate

Thoughts?

commodo, 2020-07-02:
>     * I think `_PYTHON_HOST_PLATFORM` should not be set for host pip (and perhaps even host Python), since it is installing/running code for host

fine by me

>     * For building packages for target, I'm still trying to understand if/what value of `_PYTHON_HOST_PLATFORM` is appropriate

i think i played with it a bit a few years ago, and just decided to go with linux2 for target-Python [no idea why for host-python]; there might be some auto-detection there that doesn't work well for target-Python, it could be that it felt sane to have it unified for host-python as well;

KaneGreen, 2020-07-03:
I met the same problem these days.

`pip-20.1.1` has this file `_vendor/packaging/tags.py`, while `pip-19.2.3` doesn't have.  
In fact, this file was introduced since `pip-19.3`, so `pip-19.2.3` is the last version without the file.  
Reverting `pip` to `19.2.3` may help. But it is **NOT** the ultimate solution.

BKPepe, 2020-07-05:
@jefferyto , @commodo What do you think to revert, I mean postpone update Python of 3.7 in OpenWrt 19.07 until you found workaround/solution? For Turris routers, we froze packages feed before update was introduced.

jefferyto, 2020-07-05:
I will resume working on this tomorrow, hope to find the right fix in a day or two. I would prefer to wait until later this week to decide if a revert is necessary.

commodo, 2020-07-06:
My preference would be to revert for now, and see later.
But that's because I don't have enough time to dig into it now.

jefferyto, 2020-07-10:
I believe this issue has been fixed.

KaneGreen, 2020-07-10:
My cross-compiling (on x86_64) to aarch64 succeed.

BKPepe, 2020-07-15:
Yes, I can confirm that this is fixed. Closing.

closed 2020-07-15 by BKPepe