openwrt/packages #12681: firewall: 'synflood_rate' is ignored

fantom-x, 2020-07-01:
Maintainer: @jow- 
Environment: 19.07-SNAPSHOT

Firewall is ignoring the value of `synflood_rate` and keeps using the default value of `25`:

```
        option synflood_protect '1'
        option synflood_rate '55'
        option synflood_burst '111'
```
```
reload_config
```
```
iptables-save | grep flood
:syn_flood - [0:0]
-A INPUT -p tcp -m tcp --tcp-flags FIN,SYN,RST,ACK SYN -m comment --comment "!fw3" -j syn_flood
-A syn_flood -p tcp -m tcp --tcp-flags FIN,SYN,RST,ACK SYN -m limit --limit 25/sec --limit-burst 111 -m comment --comment "!fw3" -j RETURN
-A syn_flood -m comment --comment "!fw3" -j DROP
```

feckert, 2020-07-02:
If I set the value option **synflood_rate to 55** I get the following warning on firewall reload
`Warning: Option @defaults[0].synflood_rate has invalid value '55'`
Did you check if this is a valid value?

fantom-x, 2020-07-02:
I could not find any limit, but a lot of examples online using higher values than 25. `20` is not working either.

```
Warning: Option @defaults[0].synflood_rate has invalid value '20'
Warning: Option @defaults[0].synflood_rate has invalid value '50'
```

```
limit

This module matches at a limited rate using a token bucket filter. A rule using this extension will match until this limit is reached (unless the '!' flag is used). It can be used in combination with the LOG target to give limited logging, for example.
--limit rate
Maximum average matching rate: specified as a number, with an optional '/second', '/minute', '/hour', or '/day' suffix; the default is 3/hour.
--limit-burst number
Maximum initial number of packets to match: this number gets recharged by one every time the limit specified above is not reached, up to this number; the default is 5.
```

jow-, 2020-07-02:
A suffix is mandatory. Use `option synflood_rate '55/s'`.

closed 2020-07-02 by jow-