openwrt/packages #12707: python3: python3-venv - does OpenWrt support venv? Error: Command returned non-zero exit status 1

mabachel, 2020-07-03:
Maintainer: @commodo @jefferyto
Environment: (ipq806x, Netgear R7800, OpenWrt master)

Description:
Creating a venv (`python3 -m venv .`) fails with  and Error: Command returned non-zero exit status 1
```
root@router:/tmp/pvenv# python3 -m venv .
Error: Command '['/tmp/pvenv/bin/python3', '-Im', 'ensurepip', '--upgrade', '--default-pip']' returned non-zero exit status 1.
```
I tried this as well:
```
root@router:/# python3 -m venv .
Error: Command '['/bin/python3', '-Im', 'ensurepip', '--upgrade', '--default-pip']' returned non-zero exit status 1.
```

The complete verbose output is below:
```
root@router:/tmp/pvenv# python3 -v -m venv .
import _frozen_importlib # frozen
import _imp # builtin
import '_thread' # <class '_frozen_importlib.BuiltinImporter'>
import '_warnings' # <class '_frozen_importlib.BuiltinImporter'>
import '_weakref' # <class '_frozen_importlib.BuiltinImporter'>
import '_frozen_importlib_external' # <class '_frozen_importlib.FrozenImporter'>
import '_io' # <class '_frozen_importlib.BuiltinImporter'>
import 'marshal' # <class '_frozen_importlib.BuiltinImporter'>
import 'posix' # <class '_frozen_importlib.BuiltinImporter'>
import _thread # previously loaded ('_thread')
import '_thread' # <class '_frozen_importlib.BuiltinImporter'>
import _weakref # previously loaded ('_weakref')
import '_weakref' # <class '_frozen_importlib.BuiltinImporter'>
# installing zipimport hook
import 'time' # <class '_frozen_importlib.BuiltinImporter'>
import 'zipimport' # <class '_frozen_importlib.FrozenImporter'>
# installed zipimport hook
# code object from '/usr/lib/python3.8/encodings/__init__.pyc'
# code object from '/usr/lib/python3.8/codecs.pyc'
import '_codecs' # <class '_frozen_importlib.BuiltinImporter'>
import 'codecs' # <_frozen_importlib_external.SourcelessFileLoader object at 0x450498>
# code object from '/usr/lib/python3.8/encodings/aliases.pyc'
import 'encodings.aliases' # <_frozen_importlib_external.SourcelessFileLoader object at 0x4588a8>
import 'encodings' # <_frozen_importlib_external.SourcelessFileLoader object at 0x44cae8>
# code object from '/usr/lib/python3.8/encodings/utf_8.pyc'
import 'encodings.utf_8' # <_frozen_importlib_external.SourcelessFileLoader object at 0x458018>
import '_signal' # <class '_frozen_importlib.BuiltinImporter'>
# code object from '/usr/lib/python3.8/encodings/latin_1.pyc'
import 'encodings.latin_1' # <_frozen_importlib_external.SourcelessFileLoader object at 0x4243f8>
# code object from '/usr/lib/python3.8/io.pyc'
# code object from '/usr/lib/python3.8/abc.pyc'
import '_abc' # <class '_frozen_importlib.BuiltinImporter'>
import 'abc' # <_frozen_importlib_external.SourcelessFileLoader object at 0x479028>
import 'io' # <_frozen_importlib_external.SourcelessFileLoader object at 0x4769d8>
# code object from '/usr/lib/python3.8/site.pyc'
# code object from '/usr/lib/python3.8/os.pyc'
# code object from '/usr/lib/python3.8/stat.pyc'
import '_stat' # <class '_frozen_importlib.BuiltinImporter'>
import 'stat' # <_frozen_importlib_external.SourcelessFileLoader object at 0x49ed18>
# code object from '/usr/lib/python3.8/_collections_abc.pyc'
import '_collections_abc' # <_frozen_importlib_external.SourcelessFileLoader object at 0x49ebd8>
# code object from '/usr/lib/python3.8/posixpath.pyc'
# code object from '/usr/lib/python3.8/genericpath.pyc'
import 'genericpath' # <_frozen_importlib_external.SourcelessFileLoader object at 0x450a88>
import 'posixpath' # <_frozen_importlib_external.SourcelessFileLoader object at 0x4919c8>
import 'os' # <_frozen_importlib_external.SourcelessFileLoader object at 0x48d4b8>
# code object from '/usr/lib/python3.8/_sitebuiltins.pyc'
import '_sitebuiltins' # <_frozen_importlib_external.SourcelessFileLoader object at 0x49f578>
import 'site' # <_frozen_importlib_external.SourcelessFileLoader object at 0x476488>
Python 3.8.3 (default, Jul 01 2020, 13:39:15) 
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
[... 327 lines cut ...]
# cleanup[3] wiping time
# cleanup[3] wiping _weakref
# cleanup[3] wiping _thread
# cleanup[3] wiping posix
# destroy posix
# cleanup[3] wiping marshal
# destroy marshal
# cleanup[3] wiping _io
# destroy io
# cleanup[3] wiping _warnings
# destroy _warnings
# cleanup[3] wiping _imp
# destroy _imp
# cleanup[3] wiping _frozen_importlib
# cleanup[3] wiping sys
# cleanup[3] wiping builtins
# destroy stat
# destroy genericpath
# destroy _sre
# destroy sre_parse
# destroy functools
# destroy tokenize
# destroy _heapq
# destroy _frozen_importlib
```

jefferyto, 2020-07-07:
I doubt we "officially" support venv. I've never tried it on-device and I'm not surprised if it doesn't work. (I also don't recall anyone asking about it.) Patches welcomed :smile:

commodo, 2023-01-25:
Should be fixed by  https://github.com/openwrt/packages/pull/20387

closed 2023-01-30 by commodo

sevmonster, 2023-09-28:
[This is broken on `master`].(https://github.com/commodo/packages/blob/f9403708f38b37faf8c09cef50908d908e695c45/lang/python/python3/Makefile#L190)