openwrt/packages #12708: ntpclient: stucks after a couple calls in a row, causes hotplug script to stuck, other hotplug scripts are not called anymore

MichaelUray, 2020-07-03:
@thess 

**Environment:** 
VMware
OpenWrt 19.07.3, r11063-85e04e9f46
ntpclient - 2015_365-2 - NTP client for setting system time from NTP servers.

**Description:**
Multiple calls in a row of `/usr/sbin/ntpclient -c 1 -p 123 -i 2 -h 0.openwrt.pool.ntp.org` cause the ntpclient to stuck.
This further causes that other hotplug scripts get not executet anymore, might cause security issues since it prevents the execution of firewall hotplug scripts.
I am wondering if the `ntpclient` package is still required, since an ntpclient comes with the current version of OpenWRT, perhaps the package should get removed.

Please look for more information there in the forum.
https://forum.openwrt.org/t/hotplug-iface-script-gets-not-executed-on-boot-up/67312/48

wulfy23, 2020-07-06:
while it is a valid issue... your underlying network settings and / or additional scripts / ntpclient settings played a role in the triggering this behavior. typical network settings, sane ntpclient settings and default codebase make this far less reproducible ( -5% vs your 50+% )

idfxken, 2023-01-15:
This issue is still present. Was debugging a new master build for a few hours (fritzbox with VDSL loaded from PR 11728).
No weird or special settings. Figuring out why my script wouldn´t work. 

Moving /etc/hotplug/iface/20-ntpclient and killing the waiting processes, made everything run in a batch. It is definitly reproducible on master.