openwrt/packages #12730: Openvswitch eror build lede 17.01.7

predators46, 2020-07-06:
` In file included from include/linux/random.h:10:0,
                 from /home/open/ledebuild/openwrt/build_dir/target-mips_24kc_musl-1.1.16/linux-ar71xx_generic/openvswitch-2.5.0/datapath/linux/compat/include/linux/random.h:4,
                 from include/linux/net.h:22,
                 from /home/open/ledebuild/openwrt/build_dir/target-mips_24kc_musl-1.1.16/linux-ar71xx_generic/openvswitch-2.5.0/datapath/linux/compat/include/linux/net.h:4,
                 from /home/open/ledebuild/openwrt/build_dir/target-mips_24kc_musl-1.1.16/linux-ar71xx_generic/openvswitch-2.5.0/datapath/linux/nf_conntrack_reasm.c:30:
/home/open/ledebuild/openwrt/build_dir/target-mips_24kc_musl-1.1.16/linux-ar71xx_generic/openvswitch-2.5.0/datapath/linux/nf_conntrack_reasm.c: In function 'nf_hash_frag':
/home/open/ledebuild/openwrt/build_dir/target-mips_24kc_musl-1.1.16/linux-ar71xx_generic/openvswitch-2.5.0/datapath/linux/nf_conntrack_reasm.c:80:31: error: 'struct inet_frags' has no member named 'rnd'
  net_get_random_once(&nf_frags.rnd, sizeof(nf_frags.rnd));
                               ^
include/linux/once.h:46:10: note: in definition of macro 'DO_ONCE'
     func(__VA_ARGS__);        \
          ^
include/linux/net.h:269:2: note: in expansion of macro 'get_random_once'
  get_random_once((buf), (nbytes))
  ^
/home/open/ledebuild/openwrt/build_dir/target-mips_24kc_musl-1.1.16/linux-ar71xx_generic/openvswitch-2.5.0/datapath/linux/nf_conntrack_reasm.c:80:2: note: in expansion of macro 'net_get_random_once'
  net_get_random_once(&nf_frags.rnd, sizeof(nf_frags.rnd));
  ^
/home/open/ledebuild/openwrt/build_dir/target-mips_24kc_musl-1.1.16/linux-ar71xx_generic/openvswitch-2.5.0/datapath/linux/nf_conntrack_reasm.c:80:52: error: 'struct inet_frags' has no member named 'rnd'
  net_get_random_once(&nf_frags.rnd, sizeof(nf_frags.rnd));
                                                    ^
include/linux/once.h:46:10: note: in definition of macro 'DO_ONCE'
     func(__VA_ARGS__);        \
          ^
include/linux/net.h:269:2: note: in expansion of macro 'get_random_once'
  get_random_once((buf), (nbytes))
  ^
/home/open/ledebuild/openwrt/build_dir/target-mips_24kc_musl-1.1.16/linux-ar71xx_generic/openvswitch-2.5.0/datapath/linux/nf_conntrack_reasm.c:80:2: note: in expansion of macro 'net_get_random_once'
  net_get_random_once(&nf_frags.rnd, sizeof(nf_frags.rnd));
  ^
/home/open/ledebuild/openwrt/build_dir/target-mips_24kc_musl-1.1.16/linux-ar71xx_generic/openvswitch-2.5.0/datapath/linux/nf_conntrack_reasm.c:82:33: error: 'struct inet_frags' has no member named 'rnd'
        (__force u32)id, nf_frags.rnd);
                                 ^
/home/open/ledebuild/openwrt/build_dir/target-mips_24kc_musl-1.1.16/linux-ar71xx_generic/openvswitch-2.5.0/datapath/linux/nf_conntrack_reasm.c: In function 'nf_hashfn':
/home/open/ledebuild/openwrt/build_dir/target-mips_24kc_musl-1.1.16/linux-ar71xx_generic/openvswitch-2.5.0/datapath/linux/nf_conntrack_reasm.c:94:24: error: 'const struct frag_queue' has no member named 'id'
  return nf_hash_frag(nq->id, &nq->saddr, &nq->daddr);
                        ^
/home/open/ledebuild/openwrt/build_dir/target-mips_24kc_musl-1.1.16/linux-ar71xx_generic/openvswitch-2.5.0/datapath/linux/nf_conntrack_reasm.c:94:33: error: 'const struct frag_queue' has no member named 'saddr'
  return nf_hash_frag(nq->id, &nq->saddr, &nq->daddr);
                                 ^
/home/open/ledebuild/openwrt/build_dir/target-mips_24kc_musl-1.1.16/linux-ar71xx_generic/openvswitch-2.5.0/datapath/linux/nf_conntrack_reasm.c:94:45: error: 'const struct frag_queue' has no member named 'daddr'
  return nf_hash_frag(nq->id, &nq->saddr, &nq->daddr);
                                             ^
/home/open/ledebuild/openwrt/build_dir/target-mips_24kc_musl-1.1.16/linux-ar71xx_generic/openvswitch-2.5.0/datapath/linux/nf_conntrack_reasm.c: In function 'nf_ct_frag6_expire':
/home/open/ledebuild/openwrt/build_dir/target-mips_24kc_musl-1.1.16/linux-ar71xx_generic/openvswitch-2.5.0/datapath/linux/nf_conntrack_reasm.c:111:2: error: too many arguments to function 'ip6_expire_frag_queue'
  ip6_expire_frag_queue(net, fq, &nf_frags);
  ^
In file included from /home/open/ledebuild/openwrt/build_dir/target-mips_24kc_musl-1.1.16/linux-ar71xx_generic/openvswitch-2.5.0/datapath/linux/compat/include/net/ipv6.h:6:0,
                 from /home/open/ledebuild/openwrt/build_dir/target-mips_24kc_musl-1.1.16/linux-ar71xx_generic/openvswitch-2.5.0/datapath/linux/nf_conntrack_reasm.c:43:
include/net/ipv6.h:520:6: note: declared here
 void ip6_expire_frag_queue(struct net *net, struct frag_queue *fq);
      ^
/home/open/ledebuild/openwrt/build_dir/target-mips_24kc_musl-1.1.16/linux-ar71xx_generic/openvswitch-2.5.0/datapath/linux/nf_conntrack_reasm.c: In function 'fq_find':
/home/open/ledebuild/openwrt/build_dir/target-mips_24kc_musl-1.1.16/linux-ar71xx_generic/openvswitch-2.5.0/datapath/linux/nf_conntrack_reasm.c:120:24: error: storage size of 'arg' isn't known
  struct ip6_create_arg arg;
                        ^
/home/open/ledebuild/openwrt/build_dir/target-mips_24kc_musl-1.1.16/linux-ar71xx_generic/openvswitch-2.5.0/datapath/linux/nf_conntrack_reasm.c:132:6: error: too many arguments to function 'inet_frag_find'
  q = inet_frag_find(&net->nf_frag.frags, &nf_frags, &arg, hash);
      ^
In file included from /home/open/ledebuild/openwrt/build_dir/target-mips_24kc_musl-1.1.16/linux-ar71xx_generic/openvswitch-2.5.0/datapath/linux/compat/include/net/inet_frag.h:5:0,
                 from include/net/netns/ipv4.h:9,
                 from include/net/net_namespace.h:17,
                 from /home/open/ledebuild/openwrt/build_dir/target-mips_24kc_musl-1.1.16/linux-ar71xx_generic/openvswitch-2.5.0/datapath/linux/compat/include/net/net_namespace.h:4,
                 from include/linux/netdevice.h:43,
                 from /home/open/ledebuild/openwrt/build_dir/target-mips_24kc_musl-1.1.16/linux-ar71xx_generic/openvswitch-2.5.0/datapath/linux/compat/include/linux/netdevice.h:4,
                 from /home/open/ledebuild/openwrt/build_dir/target-mips_24kc_musl-1.1.16/linux-ar71xx_generic/openvswitch-2.5.0/datapath/linux/nf_conntrack_reasm.c:32:
include/net/inet_frag.h:116:25: note: declared here
 struct inet_frag_queue *inet_frag_find(struct netns_frags *nf, void *key);
                         ^
/home/open/ledebuild/openwrt/build_dir/target-mips_24kc_musl-1.1.16/linux-ar71xx_generic/openvswitch-2.5.0/datapath/linux/nf_conntrack_reasm.c:135:3: error: implicit declaration of function 'inet_frag_maybe_warn_overflow' [-Werror=implicit-function-declaration]
[... 113 lines cut ...]
  nf_frags.match = rpl_ip6_frag_match;
          ^
/home/open/ledebuild/openwrt/build_dir/target-mips_24kc_musl-1.1.16/linux-ar71xx_generic/openvswitch-2.5.0/datapath/linux/nf_conntrack_reasm.c: In function 'nf_hash_frag':
/home/open/ledebuild/openwrt/build_dir/target-mips_24kc_musl-1.1.16/linux-ar71xx_generic/openvswitch-2.5.0/datapath/linux/nf_conntrack_reasm.c:83:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
cc1: some warnings being treated as errors
scripts/Makefile.build:277: recipe for target '/home/open/ledebuild/openwrt/build_dir/target-mips_24kc_musl-1.1.16/linux-ar71xx_generic/openvswitch-2.5.0/datapath/linux/nf_conntrack_reasm.o' failed
make[6]: *** [/home/open/ledebuild/openwrt/build_dir/target-mips_24kc_musl-1.1.16/linux-ar71xx_generic/openvswitch-2.5.0/datapath/linux/nf_conntrack_reasm.o] Error 1
Makefile:1436: recipe for target '_module_/home/open/ledebuild/openwrt/build_dir/target-mips_24kc_musl-1.1.16/linux-ar71xx_generic/openvswitch-2.5.0/datapath/linux' failed
make[5]: *** [_module_/home/open/ledebuild/openwrt/build_dir/target-mips_24kc_musl-1.1.16/linux-ar71xx_generic/openvswitch-2.5.0/datapath/linux] Error 2
make[5]: Leaving directory '/home/open/ledebuild/openwrt/build_dir/target-mips_24kc_musl-1.1.16/linux-ar71xx_generic/linux-4.4.182'
Makefile.main:71: recipe for target 'default' failed
make[4]: *** [default] Error 2
make[4]: Leaving directory '/home/open/ledebuild/openwrt/build_dir/target-mips_24kc_musl-1.1.16/linux-ar71xx_generic/openvswitch-2.5.0/datapath/linux'
Makefile:221: recipe for target '/home/open/ledebuild/openwrt/build_dir/target-mips_24kc_musl-1.1.16/linux-ar71xx_generic/openvswitch-2.5.0/.built' failed
make[3]: *** [/home/open/ledebuild/openwrt/build_dir/target-mips_24kc_musl-1.1.16/linux-ar71xx_generic/openvswitch-2.5.0/.built] Error 2
[... cut ...]
make[2]: *** [package/feeds/packages/openvswitch/compile] Error 2
make[2]: Leaving directory '/home/open/ledebuild/openwrt'
package/Makefile:101: recipe for target '/home/open/ledebuild/openwrt/staging_dir/target-mips_24kc_musl-1.1.16/stamp/.package_compile' failed
make[1]: *** [/home/open/ledebuild/openwrt/staging_dir/target-mips_24kc_musl-1.1.16/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/open/ledebuild/openwrt'
/home/open/ledebuild/openwrt/include/toplevel.mk:205: recipe for target 'world' failed `

neheb, 2020-07-06:
I don't know that LEDE 17.07 is still supported. All the buildbots have stopped.

predators46, 2020-07-07:
> Saya tidak tahu bahwa LEDE 17.07 masih didukung. Semua buildbots telah berhenti.

Thanks

closed 2020-07-07 by predators46

vaanwd, 2021-05-29:
There is a patch for this?