openwrt/packages #12731: ssmtp v2.64-7 - invalid read access from 00000000

sledz82, 2020-07-06:
Please make sure that the issue subject starts with `<package-name>: `
<package-name>: ssmtp
Maintainer: @\<Dirk Brenken>
Environment: Arch mips_24kc, Linksys WRT-160NL, OpenWrt version 19-07-SNAPSHOT

Description:

```
After configuring in /etc/ssmtp/ssmtp.conf and in /etc/ssmtp/revaliases the settings for gmail account  I have issued the following command:
cat mail.txt | sendmail -t
then Segmentation fault error occurs.

Here is the log from logread:

Sun Jul  5 20:54:49 2020 mail.info sSMTP[3442]: Creating SSL connection to host
Sun Jul  5 20:54:49 2020 mail.info sSMTP[3442]: 220 smtp.gmail.com ESMTP o9sm6951031ljh.78 - gsmtp
Sun Jul  5 20:54:49 2020 mail.info sSMTP[3442]: EHLO localhost
Sun Jul  5 20:54:49 2020 mail.info sSMTP[3442]: 250 SMTPUTF8
Sun Jul  5 20:54:49 2020 mail.info sSMTP[3442]: STARTTLS
Sun Jul  5 20:54:49 2020 mail.info sSMTP[3442]: 220 2.0.0 Ready to start TLS
Sun Jul  5 20:54:49 2020 mail.info sSMTP[3442]: SSL connection using TLS_CHACHA20_POLY1305_SHA256
Sun Jul  5 20:54:49 2020 mail.info sSMTP[3442]: EHLO localhost
Sun Jul  5 20:54:49 2020 mail.info sSMTP[3442]: 250 SMTPUTF8
Sun Jul  5 20:54:49 2020 mail.info sSMTP[3442]: AUTH LOGIN
Sun Jul  5 20:54:49 2020 mail.info sSMTP[3442]: 334 VX............
Sun Jul  5 20:54:49 2020 mail.info sSMTP[3442]: ZGF...................
Sun Jul  5 20:54:49 2020 mail.info sSMTP[3442]: 334 UG..............
Sun Jul  5 20:54:49 2020 kern.info kernel: [ 3958.182478] do_page_fault(): sending SIGSEGV to sendmail for invalid read access from 00000000
Sun Jul  5 20:54:49 2020 kern.info kernel: [ 3958.191241] epc = 77ee849c in libc.so[77e74000+98000]
Sun Jul  5 20:54:49 2020 kern.info kernel: [ 3958.196357] ra  = 004025fb in ssmtp[400000+5000]
 
```

neheb, 2020-07-06:
strace output would be useful. Also, cat is useless here. Just use < mail.txt

dibdot, 2020-07-08:
ssmtp is upstream unmaintained (since 2014). Consider using something like msmtp. I think I should remove this package here ...

dibdot, 2020-07-08:
@neheb: foolsm (currently unmaintained) depends on ssmtp, should we change that to msmtp dependency or better remove this instead?

neheb, 2020-07-08:
msmtp sounds good.

neheb, 2020-07-23:
This package was removed. Closing.

closed 2020-07-23 by neheb