openwrt/packages #12737: [19.07] openvswitch - error: 'const struct ipv6_stub'

BKPepe, 2020-07-06:
Maintainer: @yousong 
Environment: Turris 1.1, powerpc_8540 &  Turris Omnia, mvebu (cortex-a9) | OpenWrt 19.07.3. It can not be compiled.

Description:
It seems to be related to the kernel update, which was done recently in OpenWrt 19.07 branch and because of that, it can not be compiled.

```
CC [M]  /path/build/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/openvswitch-2.11.1/datapath/linux/geneve.o
/path/build/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/openvswitch-2.11.1/datapath/linux/geneve.c: In function 'geneve_get_v6_dst':
/path/build/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/openvswitch-2.11.1/datapath/linux/geneve.c:966:17: error: 'const struct ipv6_stub' has no member named 'ipv6_dst_lookup'; did you mean 'ipv6_dst_lookup_flow'?
if (ipv6_stub->ipv6_dst_lookup(geneve->net, gs6->sock->sk, &dst, fl6)) {
               ^~~~~~~~~~~~~~~
               ipv6_dst_lookup_flow
make[8]: *** [scripts/Makefile.build:326: /path/build/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/openvswitch-2.11.1/datapath/linux/geneve.o] Error 1
make[7]: *** [Makefile:1550: _module_/path/build/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/openvswitch-2.11.1/datapath/linux] Error 2
make[7]: Leaving directory '/path/build/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/linux-4.14.187'
make[6]: *** [Makefile.main:71: default] Error 2
make[6]: Leaving directory '/path/build/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/openvswitch-2.11.1/datapath/linux'
```

Also, it fails on buildbots. For example, https://downloads.openwrt.org/releases/faillogs-19.07/aarch64_cortex-a53/packages/openvswitch/compile.txt

yousong, 2020-07-14:
Two patches were backported in the bump to openvswitch 2.11.3.  Will close this issue after buildbots indicates the issue is resolved.

yousong, 2020-07-15:
The packages are now available in the downloads site.  Thanks for the report!

closed 2020-07-15 by yousong