openwrt/packages #12738: dnscrypt-proxy2: request: using upx for creating dnscrypt-proxy2

rezad1393, 2020-07-06:
Maintainer: @BKPepe 
Environment: (put here arch, model, OpenWrt version)
Linux openwrt 4.14.180 #0 SMP armv7l GNU/Linux
openwrt 19.07.03

Description:
can you use upx before packaging dnscrypt-proxy2 executable?
this way it size goes from 8 to 2.5MB.

BKPepe, 2020-07-06:
@neheb do you have any experience wih UPX for OpenWrt? I don't have it.

BKPepe, 2020-07-06:
However, go itself in some cases can be huge and I would say it expect that you have enough storage to install it and have it running, but while looking into it dnscrypt-proxy2 has 3,5 MiB.

rezad1393, 2020-07-06:
> @neheb do you have any experience wih UPX for OpenWrt? I don't have it.

I have tested upx on my desktop and used upx for amd64 and compressed the dnscrypt-proxy exe for arm extracted from ipk and then ran that on my openwrt router and it worked. (at least it complained about .toml file missing so it ran)

so I don't think you need upx for openwrt just upx for developer pc.

rezad1393, 2020-07-06:
> However, go itself in some cases can be huge and I would say it expect that you have enough storage to install it and have it running, but while looking into it dnscrypt-proxy2.ipk has 3,5 MiB.

I generally want to reduce space used by packages. the dnscrypt ipk extracted ,which would be installed on the ROM of router, produces a binary file of 8MB size.
ipk is itself compressed and that is actually not important when you create a image for installation because all the ipks get extracted and out inside the image.
also installing the ipk separately is the same: as in it would put a 8 MB binary on router.

this have some drawbacks.

one is not being able to use this on smaller ROMs. (which is not my issue but would help others)

the second drawback that is my issue is that every time I write a new image on my router I will be writing to a limited space flash (ROM) that has a limited number of writes supported (like ssd but way lower).
so every excess byte I write to it lower the lifetime of the device.

this is just a suggestion to test because this package is huge by openwrt standards so maybe doing this would also helps others too.

neheb, 2020-07-06:
Just use dnscrypt-proxy 1. Go is indeed huge.

rezad1393, 2020-07-06:
> Just use dnscrypt-proxy 1. Go is indeed huge.

that is obsolete.
plz give answers related to the issue.
that suggestion is not related to the request.

neheb, 2020-07-06:
The request will never happen.

It requires a host tool to be introduced. I also have no idea how compatible UPX is with the various platforms that OpenWrt supports.

There is also an increase in RAM usage with UPX as it needs to decompress the binary.

rezad1393, 2020-07-06:
> The request will never happen.
> 
> It requires a host tool to be introduced. I also have no idea how compatible UPX is with the various platforms that OpenWrt supports.
> 
> There is also an increase in RAM usage with UPX as it needs to decompress the binary.

that's ok.
it was just a suggestion not an order.

and adding a hoat tool doesnt destroy this project.
and testing a tool like upx (just like me) is part on the process of porgress.

also some devices like mine have more rom limit than ram limit. (512mb ran and 32mb rom). 

and the reason for my request was the rom space used by the package so that maybe the maintainer choose to do something about it or not.

neheb, 2020-07-07:
Sure but a lot of the popular routers have a RAM size of 64MB. UPX won't work well there.

As far as UPX itself, someone needs to introduce a solution ifthis is to go forward.

rezad1393, 2020-07-07:
> Sure but a lot of the popular routers have a RAM size of 64MB. UPX won't work well there.
> 
> As far as UPX itself, someone needs to introduce a solution ifthis is to go forward.

depends. they also have very small flash space.

closed 2020-07-07 by rezad1393