openwrt/packages #12741: getdns: missing dependency for libbsd.so.0

bogi788, 2020-07-07:
Maintainer: @jonathanunderwood
Environment: mips_24kc, TP-Link Archer C7 v2, OpenWRT Development Snapshot

Since #12679 getdns fails to build exactly as in #7019 due to missing its libbsd dependency:
https://downloads.openwrt.org/snapshots/faillogs/mips_24kc/packages/getdns/compile.txt

> the update seems to have brought #7019 back

Please open a new issue and provide more details including what version of OpenWRT you're using, the output you see etc.

_Originally posted by @jonathanunderwood in https://github.com/openwrt/packages/issues/12529#issuecomment-654674510_

bogi788, 2020-07-07:
#6960 fixed the problem previously

neheb, 2020-07-07:
Adding -DCMAKE_DISABLE_FIND_BSD=ON or something similar should fix it.

jonathanunderwood, 2020-07-07:
> Adding -DCMAKE_DISABLE_FIND_BSD=ON or something similar should fix it.

That's already there. I guess it's not working, and is probably the wrong flag.

bogi788, 2020-07-07:
in #6960 they used CONFIGURE_VARS += LIBBSD_LIBS=-lc

jonathanunderwood, 2020-07-07:
> in #6960 they used CONFIGURE_VARS += LIBBSD_LIBS=-lc

Yes; the move to 1.6.0 also changed the buildsystem to CMake so that fix no longer applies.

jonathanunderwood, 2020-07-07:
> Adding -DCMAKE_DISABLE_FIND_BSD=ON or something similar should fix it.

Note that CMAKE_DISABLE_FIND_XXXXX works for dependency packages that are discovered by CMake using `find_package`. But, `libbsd` searching is not done using `find_package` but `find_library`, so the -DCMAKE_DISABLE_FIND_BSD=ON is a no-op.

closed 2020-07-08 by neheb