openwrt/packages #12750: make Error

snakwu, 2020-07-08:
st_log_setup-variant-shared.cmake for lack of <pbin.v2/libs/log/build/gcc-8.4.0/release/threadapi-pthread/threading-multi/visibility-hidden>libboost_log_setup-variant-shared.cmake...
...failed updating 4 targets...
...skipped 24 targets...
...updated 10 targets...
make[3]: *** [Makefile:508: /home/snakwu/openwrt-master/openwrt/build_dir/target-x86_64_musl/boost_1_73_0/.built] Error 1
make[3]: Leaving directory '/home/snakwu/openwrt-master/openwrt/feeds/packages/libs/boost'
time: package/feeds/packages/boost/compile#123.55#147.88#263.84
make[2]: *** [package/Makefile:113: package/feeds/packages/boost/compile] Error 2
make[2]: Leaving directory '/home/snakwu/openwrt-master/openwrt'
make[1]: *** [package/Makefile:107: /home/snakwu/openwrt-master/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/snakwu/openwrt-master/openwrt'
make: *** [/home/snakwu/openwrt-master/openwrt/include/toplevel.mk:231: world] Error 2

snakwu, 2020-07-08:
dentifier could be generated
onsgmls:/home/snakwu/openwrt-master/openwrt/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:97:11: entity was defined here
onsgmls:/home/snakwu/openwrt-master/openwrt/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:339:27:E: reference to entity "gt" for which no system identifier could be generated
onsgmls:/home/snakwu/openwrt-master/openwrt/build_dir/hostpkg/expat-2.2.9/doc/xmlwf.xml:97:51: entity was defined here
onsgmls:I: maximum number of errors (200) reached; change with -E option
Can't call method "value" on an undefined value at /usr/share/perl5/sgmlspl-specs/docbook2man-spec.pl line 250, <STDIN> line 1.
make[6]: *** [CMakeFiles/expat.dir/build.make:131: libexpat.so.1.6.11] Error 8
make[6]: Leaving directory '/home/snakwu/openwrt-master/openwrt/build_dir/hostpkg/expat-2.2.9'
make[5]: *** [CMakeFiles/Makefile2:109: CMakeFiles/expat.dir/all] Error 2
make[5]: Leaving directory '/home/snakwu/openwrt-master/openwrt/build_dir/hostpkg/expat-2.2.9'
make[4]: *** [Makefile:158: all] Error 2
make[4]: Leaving directory '/home/snakwu/openwrt-master/openwrt/build_dir/hostpkg/expat-2.2.9'
make[3]: *** [Makefile:57: /home/snakwu/openwrt-master/openwrt/build_dir/hostpkg/expat-2.2.9/.built] Error 2
make[3]: Leaving directory '/home/snakwu/openwrt-master/openwrt/feeds/packages/libs/expat'
time: package/feeds/packages/expat/host-compile#0.25#0.08#0.41
make[2]: *** [package/Makefile:111: package/feeds/packages/expat/host/compile] Error 2
make[2]: Leaving directory '/home/snakwu/openwrt-master/openwrt'
make[1]: *** [package/Makefile:107: /home/snakwu/openwrt-master/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/snakwu/openwrt-master/openwrt'
make: *** [/home/snakwu/openwrt-master/openwrt/include/toplevel.mk:231: world] Error 2
snakwu@hassio-server:~/openwrt-master/openwrt$

neheb, 2020-07-08:
Cannot reproduce with both Fedora 32 and CentOS 7. What platform are you running this on?

neheb, 2020-07-08:
Fixed with https://github.com/openwrt/packages/pull/12764

closed 2020-07-08 by neheb