openwrt/packages #12769: mwan3: feature request: possibility to keep the secondary connection (e.g. pay-per-day LTE) normally down

patrakov, 2020-07-09:
Maintainer: @feckert 
Environment: Netgear R7800, master as of ~1 month ago

Description:

I have these connections:

1. Fiber WAN (through a mediaconverter), IPoE, DHCP, this is the main connection, so it is normally up
2. VPN (for censorship circumvention - there is a large ipset that directs traffic to certain IPs and networks through it), it is also normally up
3. An LTE connection, for backup if the main line fails. The protocol is NCM.

My LTE ISP charges money not for the amount of data transferred (although there is a cap), but for the fact that I had the connection up on a particular day. Therefore, it is in my interest to keep the LTE connection down if possible.

However, mwan3 does not support this setup: it requires all connections to be normally up. It adjusts routing tables, but that's it.

In my setup, this would mean the following requirements, and whatever configuration options that are required for me to be able to express them:

1. Fiber WAN and VPN should be always kept up, LTE should be down. Fiber WAN should be regularly tested by pinging a test IP.
2. If the VPN fails, try to direct traffic to the censored networks through the remaining connection, but do not bring up LTE just because of that.
3. If the fiber WAN fails (and brings the VPN down together with it), bring up LTE and direct traffic there.
4. When LTE is active and VPN connects, direct traffic to the censored networks through the VPN.
5. If the LTE connection also drops (i.e. the test IP is not pingable), try to bring it down and then up again.
6. When the main WAN becomes working again, redirect traffic through it, and bring down the LTE connection.

I understand that the logic above might be already possible with a PPP-based LTE WAN connection (dial on demand, disconnect on idle), but I have to use NCM, not PPP, because my modem supports IPv6 only via NCM.

feckert, 2020-07-09:
I think is already possible. You have to set the auto option of the LTE interface to false. So that it does not connect on boot. And if the main interface goes done, because the tracking IP´s are not reachable anymore, the the disconnected event is send. You could add a your script logic into mwan3.user and then start the LTE connection on the disconnect event. And if the main interface is connected again the you could stop the LTE connection?

patrakov, 2020-07-09:
I will check your suggestion, but I think I am not the only one with this feature request. Ideally this should be possible without scripting.

jamesmacwhite, 2020-07-16:
@patrakov I think scripting is the only way to achieve this how you want. 

One of the key features mwan3 makes use of is the hotplug events when a tracked interface status changes. You'd want to hook into these specifically for your primary WAN interface and have enable/disable logic for your LTE interface when the primary WAN is down or up. You'll have the following variables  $ACTION, $INTERFACE and $DEVICE to use to target the scenario you require.

In `/etc/mwan3.user` you'll see how these can work:

```
#!/bin/sh
#
# This file is interpreted as shell script.
# Put your custom mwan3 action here, they will
# be executed with each netifd hotplug interface event
# on interfaces for which mwan3 is enabled.
#
# There are three main environment variables that are passed to this script.
#
# $ACTION
#      <ifup>         Is called by netifd and mwan3track
#      <ifdown>       Is called by netifd and mwan3track
#      <connected>    Is only called by mwan3track if tracking was successful
#      <disconnected> Is only called by mwan3track if tracking has failed
# $INTERFACE	Name of the interface which went up or down (e.g. "wan" or "wwan")
# $DEVICE	Physical device name which interface went up or down (e.g. "eth0" or "wwan0")
```

feckert, 2020-07-18:
I think that's a great idea, but I think this is not the task of keepalived. This is more a plugin that handles this. I think you should write your own package (keepalived-connection?) that puts the script in the `/etc/hotplug.d/keepalived/00-connection` folder.  The callback script will be called at every keepalived event and then the script will be processed.

closed 2020-07-30 by feckert