openwrt/packages #12772: Stubby package not offered through opkg

speakn2pabst, 2020-07-10:
After flashing to 19.07.3 on a TP-Link Archer C7 v4 the stubby package is unavailable for installation.

After 'opkg update' and then 'opkg install stubby' through uci offers

```
Unknown package 'stubby'.
Collected errors:
 * opkg_install_cmd: Cannot install package stubby.

```

And within the Luci Software page the list of packages does not contain Stubby.

The problem seems to be confirmed by other users on other devices within the forum.
[https://forum.openwrt.org/t/installation-of-stubby-on-19-07-3-failed/68846](url)
The comments there inspired me to file this bug report, I hope this is the appropriate place.

While using the 19.07.2 series the Stubby package had been available so I'm not sure what happened.
Thanks for your time.

kyeich, 2020-07-11:
I'm using a Linksys WRT3200ACM and OpenWRT v19.07.3. I can no longer install openvswitch using opkg like I could in the past. The issue is the same as the above but for a different architecture:
* arm_cortex-a9_vfpv3-d16

The build error can be seen here:
* https://downloads.openwrt.org/releases/faillogs-19.07/arm_cortex-a9_vfpv3-d16/packages/openvswitch/

The packages for this architecture have recently been moved to "-d16" and this seems to be where the problem originated.

jonathanunderwood, 2020-07-13:
This is now fixed.

speakn2pabst, 2020-07-14:
I've attempted installing stubby and it was indeed fixed.  Many thanks to those involved, the OpenWRT contributors and community.

closed 2020-07-14 by speakn2pabst