openwrt/packages #12774: numpy: Build error for arc770

jefferyto, 2020-07-10:
Maintainer: @commodo 
Environment: arc770, master

Description:

https://downloads.openwrt.org/snapshots/faillogs/arc_arc700/packages/numpy/compile.txt

I think this is the relevant error message:

```
numpy/core/src/multiarray/multiarraymodule.c: In function 'array_vdot':
numpy/core/src/multiarray/multiarraymodule.c:2358:1: error: unable to generate reloads for:
 }
 ^
(jump_insn 165 284 166 13 (parallel [
            (set (pc)
                (unspec:SI [
                        (reg:SI 17 r17 [203])
                    ] UNSPEC_ARC_CASESI))
            (use (label_ref 166))
            (clobber (scratch:SI))
        ]) "numpy/core/src/multiarray/multiarraymodule.c":2319 149 {casesi_compact_jump}
     (expr_list:REG_DEAD (reg:SI 17 r17 [203])
        (nil))
 -> 166)
during RTL pass: reload
numpy/core/src/multiarray/multiarraymodule.c:2358:1: internal compiler error: in find_reloads, at reload.c:3814
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://bugs.openwrt.org/> for instructions.
```

There is also a similar error further down:

```
numpy/core/src/multiarray/lowlevel_strided_loops.c.src: In function 'PyArray_GetStridedCopyFn':
numpy/core/src/multiarray/lowlevel_strided_loops.c.src:476:1: error: unable to generate reloads for:
 }
 ^
(jump_insn 71 344 72 7 (parallel [
            (set (pc)
                (unspec:SI [
                        (reg:SI 3 r3 [204])
                    ] UNSPEC_ARC_CASESI))
            (use (label_ref 72))
            (clobber (scratch:SI))
        ]) "numpy/core/src/multiarray/lowlevel_strided_loops.c.src":350 149 {casesi_compact_jump}
     (expr_list:REG_DEAD (reg:SI 3 r3 [204])
        (nil))
 -> 72)
during RTL pass: reload
numpy/core/src/multiarray/lowlevel_strided_loops.c.src:476:1: internal compiler error: in find_reloads, at reload.c:3814
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://bugs.openwrt.org/> for instructions.
```

commodo, 2020-07-13:
ohhh; that looks like a compiler bug;

commodo, 2020-07-13:
opened https://bugs.openwrt.org/index.php?do=details&task_id=3232
maybe we can filter out this arch in the depends section;

objections?

jefferyto, 2020-07-13:
> Please submit a full bug report,
> with preprocessed source if appropriate.
> See <http://bugs.openwrt.org/> for instructions.

Is there any "preprocessed source" that should be submitted to FS#3232?

Also, looking at FS#3232 it doesn't seem immediately clear to me that this is a main repo issue and not a packages repo issue.

sielicki, 2020-07-13:
From gcc's ChangeLog-2018

```
 6471 2018-10-31  Claudiu Zissulescu  <claziss@synopsys.com>
 6472 
 6473 	* config/arc/arc.c (arc_override_options): Remove
 6474 	TARGET_COMPACT_CASESI.
 6475 	* config/arc/arc.h (ASM_OUTPUT_ADDR_DIFF_ELT): Update.
 6476 	(CASE_VECTOR_MODE): Likewise.
 6477 	(CASE_VECTOR_PC_RELATIVE): Likewise.
 6478 	(CASE_VECTOR_SHORTEN_MODE): Likewise.
 6479 	(CASE_VECTOR_SHORTEN_MODE1): Delete.
 6480 	(ADDR_VEC_ALIGN): Update.
 6481 	(ASM_OUTPUT_CASE_LABEL): Undefine.
 6482 	(ASM_OUTPUT_BEFORE_CASE_LABEL): Undefine.
 6483 	(TARGET_BI_BIH): Define.
 6484 	(DEFAULT_BRANCH_INDEX): Likewise.
 6485 	* config/arc/arc.md (casesi): Rework to accept BI/BIH
 6486 	instructions, remove compact_casesi use case.
 6487 	(casesi_compact_jump): Remove.
 6488 	(casesi_dispatch): New pattern.
 6489 	* config/arc/arc.opt: Add mbranch-index option. Deprecate
 6490 	compact_casesi option.
 6491 	* doc/invoke.texi: Document mbranch-index option.
```

note line 6487: `casesi_compact_jump` was removed. See: https://gcc.gnu.org/legacy-ml/gcc-patches/2018-10/msg00539.html

Fixing this bug without bumping GCC probably means backporting these machine description changes? Yuck.

commodo, 2020-07-14:
well, if backporting just that patch would work, then i'd be fine to try that;
thanks for isolating this patch;
will try it

commodo, 2020-07-15:
so, backporting that patch helps;
the build failure is gone;
this patch looks present in GCC 9, I backported it to GCC 8

commodo, 2020-07-15:
create PR https://github.com/openwrt/openwrt/pull/3208 for it;

closed 2025-11-08 by GeorgeSapkin