openwrt/packages #12776: nginx-util: how to use different port

jkcapoeira, 2020-07-10:
Maintainer: @Ansuel @peter-stadler
Environment: OpenWrt snapshot

Description:
I am using sslh to listen on port 443 and split the traffic into 9443 for nginx and 1194 for openvpn. Everytime nginx restarts, it creates the file lan_ssl.listen which is also on port 443?

peter-stadler, 2020-07-10:
This is true, but per default this file is included in `/etc/nginx/conf.d/_lan.conf` only.

For your setup, replace the `_lan.conf` with an empty file or remove at least the corresponding include directives from it (do not delete the file, opkg would reinstall it on an upgrade). Then the default server (on port 443) is disabled.

You can create a server listening on port 9443 in a file `/etc/nginx/conf.d/ssh.conf` for example containing something like:
```
server {
     listen 9443;
     …
}
```
All `/etc/nginx/conf.d/*.conf` files are included in the `http` part of the main config. Looking at [the discussion](https://forum.openwrt.org/t/nginx-and-openvpn-on-port-443/25004/4) I am not sure if the server part has to go into a `stream` part instead; in that case you have to edit the main config directly.

More details are in the [wiki](https://openwrt.org/docs/guide-user/services/webserver/nginx#new_server_parts). I will update the infos there, when we use the new version of [nginx-util](https://github.com/openwrt/packages/pull/11456) …

peter-stadler, 2020-07-10:
If you want the default server to listen on another port without disabling it, you can replace the directive `include '/var/lib/nginx/lan_ssl.listen.default';` in the file `/etc/nginx/conf.d/_lan.conf` by `listen 9443 ssl default_server;` for example (I am not sure if you want to do that, but maybe others who get here reading the title of the issue).