openwrt/packages #12777: collectd-mod-ntpd requires change in /var/etc/ntpd.conf

mnlipp, 2020-07-10:
Maintainer: @feckert (CC: @tripolar)
Environment: OpenWrt 19.07.03

Description:

The collectd ntpd plugin uses the "private mode" of ntpd to communicate with ntpd. This mode has been deprecated and must be enabled explicitly (see [here](https://github.com/collectd/collectd/issues/932)).

The `ntpd.conf` is generated by `/etc/init.d/ntpd` which does not provide a way to add lines to the generated `/var/etc/ntpd.conf`. I have added the required line explicitly (`emit "enable mode7"`).

Of course, this is a hack. What we require here is a possibility to add lines to the `ntpd.conf` "triggered" by the installation of another package (e.g. collectd-mod-ntpd).

feckert, 2020-07-13:
First, I think we should teach ntpd to load additional configuration. The documentation includes following config option. I could not check this because i am using the busybox ntp. But maybe this works? I did not do a run test.

> includefile includefile
>            This   command  allows  additional  configuration  commands  to  be
>            included from a separate file. Include files may  be  nested  to  a
>            depth  of  five; upon reaching the end of any include file, command
>            processing resumes in the previous configuration file. This  option
>            is  useful for sites that run ntpd on multiple hosts, with (mostly)
>            common options (e.g., a restriction list). The include file must be
>            in a file readable by the user "daemon".

We could add the following lines to the ntpd.init script.

```
diff --git a/net/ntpd/files/ntpd.init b/net/ntpd/files/ntpd.init
index 0f0f57700..a8bb7b3cd 100644
--- a/net/ntpd/files/ntpd.init
+++ b/net/ntpd/files/ntpd.init
@@ -75,6 +75,13 @@ start_ntpd_instance() {
                emit "server $i iburst"
        done

+       if [ -d "/tmp/ntpd.d" ]; then
+               local entry
+               for entry in "/tmp/ntpd.d"*; do
+                       emit "includefile ${entry}\n"
+               done
+       fi
+
        mkdir -p /var/lib/ntp
        chown -R ntp:ntp /var/lib/ntp
```

The collectd could then place the options in a file under `/tmp/ntpd.d/01-collectd-mod-ntpd` and hooks the reload of `/etc/init.d/ntpd`. The reload hook reads the config option in this file.

mnlipp, 2020-07-14:
There's a slash missing and why would you put the to be included files in `/tmp`? I'd use `/etc` as usual to also allow additional "manually maintained" additional configuration (similar to `/etc/collectd/conf.d/`). 

In general, I prefer to restrict wildcards using an extension. This allows you to leave e.g. a `README` in the directory.

Here's my tested code:
```
        if [ -d "/etc/ntpd.d" ]; then
                local entry                      
                for entry in "/etc/ntpd.d"/*.conf; do
                       emit "includefile ${entry}\n"      
                done
        fi 
```

feckert, 2020-07-14:
> There's a slash missing and why would you put the to be included files in `/tmp`? I'd use `/etc` as usual to also allow additional "manually maintained" additional configuration (similar to `/etc/collectd/conf.d/`).

Sure we could do this as well.
But I thought if we dynamically create configurations, for example from uci, I think it would be better to do it in tmp.
Because otherwise we have a lot of write cycles.

> In general, I prefer to restrict wildcards using an extension. This allows you to leave e.g. a `README` in the directory.

You're right, of course. It was just written very quickly.

mnlipp, 2020-07-14:
> But I thought if we dynamically create configurations, for example from uci, I think it would be better to do it in tmp.

For this particular use case, configuration could be "static". As `collectd-mod-ntpd` works only with `enable mode7` in the ntpd configuration, the package for `collectd-mod-ntpd` could simply install the file `/etc/ntpd.d/01-collectd-mod-ntpd.conf` with content `enable mode7`. No need to make things unnecessarily complicated.

If @tripolar wants to, he could add the section above for both `/etc/ntpd.d` and `/tmp/ntpd.d`, although we currently have no exinsting use case for the latter (thus being "future-ready").

neheb, 2020-09-19:
The PR was merged.

closed 2020-09-19 by neheb

mnlipp, 2021-05-14:
I've just upgraded to 19.07.07 and wonder, what the PR was merged with. Because this is still missing...