openwrt/packages #12793: docker-ce: Makefile:380: *** go-bootstrap cannot be installed on linux/arm64.  Stop.

alex9434, 2020-07-12:
Please make sure that the issue subject starts with `<package-name>: `

Also make sure that the package is maintained in this repository and not in base which should be submitted at https://bugs.openwrt.org or in the LuCI repository which should be submitted at https://github.com/openwrt/luci/issues.

Issues related to releases below 18.06 and forks are not supported or maintained and will be closed.

# Issue template (remove lines from top till here)

Maintainer: @G-M0N3Y-2503 @feckert @jneuhauser @jefferyto @jmarcet @Student414 
Environment: aarch64, Raspberry Pi 4B, Snapshot
Build system: Linux debian-buster-64 4.19.118-v8+ #1311 SMP PREEMPT Mon Apr 27 14:32:38 BST 2020 aarch64 GNU/Linux

Description:
Package docker-ce does not build on Debian 10 Aarch64. error message:
```
docker-ce: Makefile:380: *** go-bootstrap cannot be installed on linux/arm64.  Stop.
```

jefferyto, 2020-07-12:
1. Install Go on your build system, either using your distro’s packages or from golang.org.
2. In `make menuconfig`, find the “ External bootstrap Go root directory” option for the golang package and enter the path for the working Go tree installed in the previous step.
3. Build as normal.

jefferyto, 2020-07-13:
If you installed Go using the Debian package, then the working Go tree path would be `/usr/lib/go-1.11` (or `/usr/lib/go-1.14` if you are using buster-backports).

The instructions I provided should have resolved your issue. Please comment if you are able to build Go/Docker successfully or if you have encountered additional issues (or if the instructions were unclear).

Also, in the future please ping (@ mention) only the maintainer of the package (listed in the package's Makefile), not every contributor who has committed a change for the package. Pinging everyone does not lead to your issue being resolved faster, it leads to people being less willing to help you.

alex9434, 2020-07-14:
Hi @jefferyto, I am just trying to compile with your suggestions. Thank you for your help and the very quick answer! Really appreciated. Unfortunately I just run into another issue with runc: 
`openwrt/build_dir/target-aarch64_cortex-a72_musl/runc-1.0.0-rc10/.go_work/build/src/github.com/opencontainers/runc: No such file or directory.  Stop.`
Would you happen to know how to fix that? I am trying to compile docker-ce from openwrt-master. 
Apologies for tagging everybody. I was just following the instruction from the ticket that asked to include contributors from the makefile. I was not trying to be 'pushy'. Will keep that in mind moving forward.

jefferyto, 2020-07-15:
That looks like an issue with the runc package. Can you create a new issue for runc and include the full build/make log?

alex9434, 2020-07-15:
Hi @jefferyto I have seen that you are also maintainer of rung. I solved the issue above with the following recommendation:

https://forum.openwrt.org/t/runc-and-containerd-compilation-fails/60572

(comment out #rm -f $(STAMP_BUILT). Just as an FYI in case you want to fix that upstream. Thank you again for your support.

closed 2020-07-15 by alex9434

jefferyto, 2020-07-15:
I'm glad you were able to find a fix.

Just to clear things up: I am not the maintainer of runc; the maintainer is Gerard Ryan. You need to look inside the package [Makefile](https://github.com/openwrt/packages/blob/335efed8cb56bcc2c140efcb42a3b70fb4b4a14c/utils/runc/Makefile#L14) to find the maintainer's name. The comment in the issue template tells you how to find the GitHub username for the maintainer.

geopenwu, 2023-12-18:
> 1. Install Go on your build system, either using your distro’s packages or from golang.org.
> 2. In `make menuconfig`, find the “ External bootstrap Go root directory” option for the golang package and enter the path for the working Go tree installed in the previous step.
> 3. Build as normal.

Thank you! I had the same issue, and resolved it with this instruction.