openwrt/packages #12802: Some packages depends in telephony are not follow syntax

ElonH, 2020-07-13:
I found some packages depends looks weird in `tmp/.packageinfo`

Example:
```

Package: baresip-mod-pulse
Submenu: Telephony
Version: 0.6.4-2
Depends: +libc +USE_GLIBC:librt +USE_GLIBC:libpthread baresip PACKAGE_baresip-mod-pulse:pulseaudio
Conflicts: 
Menu-Depends: 

```

`PACKAGE_baresip-mod-pulse:pulseaudio` doesn't follow any rule in <https://openwrt.org/docs/guide-developer/dependencies> or <https://openwrt.org/docs/guide-developer/packages>, missing `@`.

Does it not raises any problem?

Other packages like above:
 - libfreetdm-ftmod-analog
 - libfreetdm-ftmod-analog-em
 - libfreetdm-ftmod-libpri
 - libfreetdm-ftmod-pritap
 - libfreetdm-ftmod-skel
 - libfreetdm-ftmod-zt

> All of these packages depends are automatically generated

micmac1, 2020-07-13:
Hi there,

This is not the telephony repo. You are in packages.

And these kinds of dependencies _are_ mentioned in the first link you quoted. See

`DEPENDS:=+PACKAGE_arpd:libpcap`

This may look a bit strange, but it's a way to avoid pulling in dependencies you don't need. And I mean usually for modular packages.

Imagine you're interested in baresip, but you're not interested in baresip-mod-pulse, so you don't select the latter in menuconfig. So you'd think that pulseaudio doesn't get pulled in, right? Well, but it will be pulled in, except when you put the dependency like we did.

Kind regards,
Seb

micmac1, 2020-07-13:
I wasn't being clear. I meant build-time deps. All this doesn't change runtime-deps.

ElonH, 2020-07-14:
@micmac1 

I knew the rules like `DEPENDS:=@symbol:pack` and `DEPENDS:=+symbol:package` are right.

But if the rule `DEPENDS:=symbol:package` is right (without prefix `@` or `+`)?

Because some depends item in mentioned pacakges are missing prefix `@` or `+`.

micmac1, 2020-07-14:
I can't explain it better than the page that you already linked. https://openwrt.org/docs/guide-developer/dependencies

closed 2020-07-23 by ElonH